BreeeZe/rpos #4: onvif client sends GetProfiles producing error in onvif server

mquevedob, 2016-01-25:
Hi, 

I have an IP camera that uses ONVIF protocol. I am running your ONVIF server and when the IP camera sends GetProfiles the ONVIF server throws this error.

```
DeviceService - Calltype : error, Data : TypeError: Cannot read property 'output' of undefined
    at Object.p.onopentag (/home/pi/rpos-0.1.0/node_modules/soap/lib/wsdl.js:1255:40)
    at emit (/home/pi/rpos-0.1.0/node_modules/soap/node_modules/sax/lib/sax.js:639:35)
    at emitNode (/home/pi/rpos-0.1.0/node_modules/soap/node_modules/sax/lib/sax.js:644:5)
    at openTag (/home/pi/rpos-0.1.0/node_modules/soap/node_modules/sax/lib/sax.js:839:5)
    at Object.write (/home/pi/rpos-0.1.0/node_modules/soap/node_modules/sax/lib/sax.js:1278:15)
    at WSDL.xmlToObject (/home/pi/rpos-0.1.0/node_modules/soap/lib/wsdl.js:1381:5)
    at Server._process (/home/pi/rpos-0.1.0/node_modules/soap/lib/server.js:151:21)
    at IncomingMessage.<anonymous> (/home/pi/rpos-0.1.0/node_modules/soap/lib/server.js:124:14)
    at emitNone (events.js:67:13)
    at IncomingMessage.emit (events.js:166:7)
```

I would be great if you can give me some pointers how to solve this.

Regards,
Martin

mquevedob, 2016-01-25:
further investigation on the matter showed me that my IP camera is sending this request to device_service, but your server expects it on media_service, I think.

Any suggestions on how to solve this easily?

RogerHardiman, 2016-04-18:
When you write "IP Camera" I think you mean "IP Camera Viewing Software or ONVIF Client".

RPOS has a SOAP GetCapabilities function that tells the ONVIF Client (the viewing software) where to send 'media' related SOAP messages (like GetProfiles) to.
If the ONVIF Client (the viewing software) is sending GetProfiles to device_service, then there is a bug in your Camera Viewing Software (or ONVIF client)

RogerHardiman, 2016-07-07:
Can you send any more information. For example what software were you using to connect to RPOS

RogerHardiman, 2017-05-14:
Closing this one down. We never received any more information on the problem.

closed 2017-05-14 by RogerHardiman