BreeeZe/rpos #5: search with onvif device manager

doanthaibao, 2016-02-19:
It maybe not an issue. 
Could rpos support search via onvif device manager?
Thanks
Bao

RogerHardiman, 2016-04-18:
I am working on ONVIF Device Manager support

I don't think I will have time to add full discovery but you can do a Manual Add to ONVIF Device Manager.
Right now ONVIF Device Manager fails with RPOS due to some RPOS bugs

Bug Fix 1
The SOAP envelope returned by RPOS is SOAP1.1. It needs to be SOAP1.2 and so needs an edit to the soap library server.js (see my another Issue Report)

Bug Fix 2
In media_service.js the timeout is not in the correct format.
Change the Timeout from 1000 to PT1000S

This is enough to make ONVIF Device Manager connect to RPOS. But it does not show a list of profiles yet

(I will add more bug fixes later)

RogerHardiman, 2016-04-18:
Latest update. I can now access RPOS from ONVIF Device Manager.
I had to fix a few incorrect SOAP messages and add a new unimplemented SOAP messages.
But now ONVIF Device Manager can connect to the Raspberry Pi, show hardware information and Stream Live Video.

doanthaibao, 2016-04-19:
Thank for your support. I will check it on my Pi.

RogerHardiman, 2016-04-19:
Hi
I have made some local changes to add extra ONVIF messages and fix sone invalid XML so let me know how you get on.
Roger

doanthaibao, 2016-04-19:
I changed some xml
`//  "<soap:Envelope xmlns:soap=\"http://schemas.xmlsoap.org/soap/envelope/\" " +
    "<soap:Envelope xmlns:soap=\"http://www.w3.org/2003/05/soap-envelope\" " +`
Onvif device manage can receive Pi info but can't stream live video. Do you have any ideas or I will wait for your change as you mentioned.
[image]
Regards
Bao

RogerHardiman, 2016-04-22:
RPOS was missing some important ONVIF message, mainly GetScopes and GetVideoSources.
I have made a fork with the fixes in (I have only edited the .ts files so you will need to compile the javascript yourself

BreeeZe, 2016-04-29:
I merged the changes from @RogerHardiman to the repo.

closed 2016-04-29 by BreeeZe