BreeeZe/rpos #8: RTSP server : image quality

doanthaibao, 2016-04-19:
I use VLC or some DVR (for example https://geniusvision.net/ ) test on your software and I see that the quality isn't good, sometimes video is delayed.
I tested with another rtsp on PI https://github.com/mpromonet/h264_v4l2_rtspserver
with video 720/1280 (15 to 30fps) it seems better (no delay)
I suggest that maybe you should switch to this library or do you have any idea?
(In my opinion, you called execution file from nodejs)
I'm researching a library support RTSP, ONVIF, discovery for Raspberry to build a smart camera from Pi, the final target is integrating Raspberry smart camera to any DVRs or NVRs.
I hope get some advises from you.
Regards 
Bao

RogerHardiman, 2016-04-19:
I noticed that by default the bitrate (and therefore the image quality) 
is very low with this software by default.
I don't know where the default settings are but I was able to change the 
image quality with ONVIF Device Manager (with my local changes that I 
have not added to Git yet) and when I increased the bitrate the picture 
was excellent.

doanthaibao, 2016-04-20:
Hi Roger,
Sorry about the title isn't correct, I correct the title. I mean that sometimes the video delay and don't capture image anymore. 
Thanks
Bao

BlueT, 2016-05-11:
@doanthaibao for the video delay, what bitrate/fps/resolution are you using?
And how about the CPU/Memory usage?

Seems like a performance issue