BreeeZe/rpos #10: Added suppport for USB Camera

Jalict, 2016-05-05:
I just realised that this didn't use USB camera. As right now I get
```
child_process.js:484
    throw err;
    ^

Error: Command failed: sudo v4l2-ctl --set-fmt-video=pixelformat=4

    at checkExecSyncError (child_process.js:441:13)
    at Object.execSync (child_process.js:481:13)
    at Function.utils.execSync (/home/pi/rpos-0.1.0/lib/utils.js:94:66)
    at execV4l2 (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:173:22)
    at Object.SetPixelFormat (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:277:9)
    at Camera.setupCamera (/home/pi/rpos-0.1.0/lib/camera.js:129:27)
    at new Camera (/home/pi/rpos-0.1.0/lib/camera.js:44:14)
    at Object.<anonymous> (/home/pi/rpos-0.1.0/rpos.js:23:14)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
```

Any possibility that this could be added? It is apparantly not common for people to use USB Cameras as quick security cameras and everyone just uses the camera modules :frowning:

RogerHardiman, 2016-05-05:
Hi Jalict
RPOS requires a Video4Linux camera that can deliver H264 video - ie video that is already compressed in H264 format   (Pixel Format 4 means H64 format)

There is a 'virtual Video4Linux driver that can take a USB camera and compress the video with linux264 to give H264 video. I have not tested it yet but I think it will work.
There is some information on this page
[https://github.com/mpromonet/h264_v4l2_rtspserver](https://github.com/mpromonet/h264_v4l2_rtspserver)

Using the example on this web site they create a new V4L driver (/dev/video10) and you will need to make a small change to RPOS to look for /dev/video10 instead of /dev/video0

I was thinking of adding in support for the H264_v4l2_rtspserver code in RPOS but have run out of development time.

jgiesen, 2018-03-06:
Hi, sorry for reopening this, but i can't get it to work:
I tried the suggested v4l2loopback, adapted the "ros/lib/camera.js" by changing the 3 video0 to video9, but rpos does not give video output.
i don't know where to search, because almost everything runswithout errors:

v4l2loopback: 
> pi@raspberrypi:~ $ sudo  modprobe v4l2loopback video_nr=9
> pi@raspberrypi:~ $ sudo  v4l2compress_h264 /dev/video0 /dev/video9 &
> [1] 1018
> pi@raspberrypi:~ $ 2018-03-06 15:21:56,403 [NOTICE] - src/V4l2Device.cpp:117
> 	driver:uvcvideo capabilities:84200001 mandatory:4000001
> 2018-03-06 15:21:56,404 [NOTICE] - src/V4l2Device.cpp:120
> 	/dev/video0 support capture
> 2018-03-06 15:21:56,404 [NOTICE] - src/V4l2Device.cpp:123
> 	/dev/video0 support streaming
> 2018-03-06 15:21:56,404 [NOTICE] - src/V4l2Device.cpp:147
> 	/dev/video0:YUYV size:640x360
> 2018-03-06 15:21:56,406 [NOTICE] - src/V4l2Device.cpp:208
> 	/dev/video0:YUYV size:640x480 bufferSize:614400
> 2018-03-06 15:21:56,408 [NOTICE] - src/V4l2Device.cpp:229
> 	fps:1/25
> 2018-03-06 15:21:56,408 [NOTICE] - src/V4l2Device.cpp:230
> 	nbBuffer:0
> 2018-03-06 15:21:56,415 [NOTICE] - src/V4l2MmapDevice.cpp:71
> 	Device /dev/video0 nb buffer:10
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:117
> 	driver:v4l2 loopback capabilities:85208003 mandatory:4000002
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:119
> 	/dev/video9 support output
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:120
> 	/dev/video9 support capture
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:122
> 	/dev/video9 support read/write
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:123
> 	/dev/video9 support streaming
> 2018-03-06 15:21:56,420 [NOTICE] - src/V4l2Device.cpp:147
> 	/dev/video9:BGR4 size:0x0
> 2018-03-06 15:21:56,423 [NOTICE] - src/V4l2Device.cpp:208
> 	/dev/video9:H264 size:640x480 bufferSize:1228800
> 2018-03-06 15:21:56,423 [NOTICE] - src/V4l2MmapDevice.cpp:71
> 	Device /dev/video9 nb buffer:8
> 2018-03-06 15:21:56,427 [NOTICE] - src/v4l2compress_h264.cpp:136
> 	Start Capturing from /dev/video0
> 2018-03-06 15:21:56,441 [WARN  ] - src/v4l2compress_h264.cpp:163
> 	rc_method:1
> 2018-03-06 15:21:56,442 [WARN  ] - src/v4l2compress_h264.cpp:164
> 	i_qp_constant:23
> 2018-03-06 15:21:56,442 [WARN  ] - src/v4l2compress_h264.cpp:165
> 	f_rf_constant:23
> x264 [info]: using cpu capabilities: ARMv6 NEON
> x264 [info]: profile Constrained Baseline, level 3.0
> 2018-03-06 15:21:56,569 [NOTICE] - src/v4l2compress_h264.cpp:185
> 	Start Compressing /dev/video0 to /dev/video9

and rpos tells me:

> pi@raspberrypi:~/rpos $ sudo node rpos.js
> Read IP address 192.168.25.128 from eth0
> Manufacturer : Raspberry Pi
> Model : 2 B
> HardwareId : 
> SerialNumber : 0000000068e54c90
> FirmwareVersion : 2.0.0
> Starting camera settings webserver on http://192.168.25.128:8081/
> Could not retrieve Controlvalue 'red_balance'
> Could not retrieve Controlvalue 'blue_balance'
> Could not retrieve Controlvalue 'horizontal_flip'
> Could not retrieve Controlvalue 'vertical_flip'
> Could not retrieve Controlvalue 'color_effects'
> Could not retrieve Controlvalue 'rotate'
> Could not retrieve Controlvalue 'color_effects_cbcr'
> Could not retrieve Controlvalue 'video_bitrate_mode'
> Could not retrieve Controlvalue 'video_bitrate'
> Could not retrieve Controlvalue 'repeat_sequence_header'
> Could not retrieve Controlvalue 'h264_i_frame_period'
> Could not retrieve Controlvalue 'h264_level'
> Could not retrieve Controlvalue 'h264_profile'
> Could not retrieve Controlvalue 'auto_exposure'
> Could not retrieve Controlvalue 'exposure_time_absolute'
> Could not retrieve Controlvalue 'exposure_dynamic_framerate'
> Could not retrieve Controlvalue 'auto_exposure_bias'
> Could not retrieve Controlvalue 'white_balance_auto_preset'
> Could not retrieve Controlvalue 'image_stabilization'
> Could not retrieve Controlvalue 'iso_sensitivity'
> Could not retrieve Controlvalue 'exposure_metering_mode'
> Could not retrieve Controlvalue 'scene_mode'
> Could not retrieve Controlvalue 'compression_quality'
> Binding DeviceService to http://192.168.25.128:8081/onvif/device_service
> Binding MediaService to http://192.168.25.128:8081/onvif/media_service
> Binding PTZService to http://192.168.25.128:8081/onvif/ptz_service
> discovery_service started
> Starting Live555 rtsp server
> media_service started
> rtspServer error: Error: spawn v4l2rtspserver ENOENT
> ptz_service started
> device_service started

so i guess rposdoesn't like the output of v4l2loopback... please help

Thanks!

13030205643, 2018-11-16:
hey! I want to know will you added suppport for USB Camera ???

RogerHardiman, 2018-12-30:
Hi @jiayaoO3O 
There are plans to add USB camera support, probably using the new GStreamer RTSP server that has recently been added.
However it could be a long time before I get to it, so if you want to look at the GStreamer code and make a new gst-launch pipeline with x264enc that would be great