BreeeZe/rpos #13: Added gulp tasks and migrated to TypeScript 1.8

BreeeZe, 2016-05-10:
@RogerHardiman I created a new branch with some refactors: https://github.com/BreeeZe/rpos/tree/dev
Can you have a look if works on your end?
Maybe you need to do a clean checkout and just run "npm install" and after that run "gulp" to run the default gulp task which compiles the typescript.

I will have some time to explain some more tomorrow if need, I'll probably add a segment to the README.MD file about development.

RogerHardiman, 2016-05-10:
Hi
Thanks for tiding up the discovery_service and getting rid of the 
string.Replace (xxx,yyy) code. The .Replace() was a quick way to get 
something working and I ran out of time to go back and make that code 
cleaner. Glad you were able to improve it.

I noticed package.json includes   "uuid5": "1.0.1" but that is not 
needed now that you have added it to utils.ts

I have just done a fresh clone and checkout of the dev branch.
"npm install" is taking a very long time on my Pi3  (over 30 minutes so 
far) so will report tomorrow.

RogerHardiman, 2016-05-10:
Hi
"gulp" completes and gives me .js files which I can then run with "node 
rpos.js"
It looks OK and the two pieces of CCTV viewing software I am using can 
discover RPOS on the LAN and view the live view.

[21:31:52] Using gulpfile ~/rpos-dev/rpos/gulpfile.js
[21:31:52] Starting 'default'...
[21:31:52] Starting 'typings'...
[21:31:52] Finished 'typings' after 5.71 ms
[21:31:52] Starting 'compile'...
lib/SoapService.ts(1,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
lib/SoapService.ts(4,21): error TS2307: Cannot find module 'fs'.
lib/SoapService.ts(6,24): error TS2307: Cannot find module 'http'.
lib/SoapService.ts(7,17): error TS2304: Cannot find name 'require'.
lib/camera.ts(1,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
lib/camera.ts(4,21): error TS2307: Cannot find module 'fs'.
lib/camera.ts(5,25): error TS2307: Cannot find module 'body-parser'.
lib/camera.ts(6,30): error TS2307: Cannot find module 'child_process'.
lib/utils.ts(1,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
lib/utils.ts(2,35): error TS2307: Cannot find module 'os'.
lib/utils.ts(3,37): error TS2307: Cannot find module 'child_process'.
lib/utils.ts(4,29): error TS2307: Cannot find module 'events'.
lib/utils.ts(5,36): error TS2307: Cannot find module 'stream'.
lib/utils.ts(6,25): error TS2307: Cannot find module 'crypto'.
lib/utils.ts(8,11): error TS2304: Cannot find name 'require'.
lib/utils.ts(94,26): error TS2304: Cannot find name 'process'.
lib/utils.ts(94,62): error TS2304: Cannot find name 'process'.
lib/utils.ts(124,35): error TS2304: Cannot find name 'require'.
lib/utils.ts(141,7): error TS2304: Cannot find name 'process'.
lib/utils.ts(144,7): error TS2304: Cannot find name 'process'.
lib/utils.ts(145,9): error TS2304: Cannot find name 'process'.
lib/utils.ts(149,7): error TS2304: Cannot find name 'process'.
lib/utils.ts(151,9): error TS2304: Cannot find name 'process'.
lib/utils.ts(155,7): error TS2304: Cannot find name 'process'.
lib/utils.ts(157,9): error TS2304: Cannot find name 'process'.
lib/v4l2ctl.ts(2,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
lib/v4l2ctl.ts(4,45): error TS2307: Cannot find module 'fs'.
rpos.ts(2,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
rpos.ts(27,1): error TS2304: Cannot find name 'require'.
rpos.ts(29,23): error TS2307: Cannot find module 'http'.
rpos.ts(30,26): error TS2307: Cannot find module 'express'.
rpos.ts(38,13): error TS2304: Cannot find name 'require'.
rpos.ts(39,26): error TS2304: Cannot find name 'require'.
services/device_service.ts(1,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
services/device_service.ts(4,21): error TS2307: Cannot find module 'fs'.
services/device_service.ts(5,23): error TS2307: Cannot find module 'util'.
services/device_service.ts(6,21): error TS2307: Cannot find module 'os'.
services/device_service.ts(9,24): error TS2307: Cannot find module 'http'.
services/device_service.ts(17,27): error TS2304: Cannot find name 'require'.
services/discovery_service.ts(2,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
services/discovery_service.ts(31,24): error TS2307: Cannot find module 
'dgram'.
services/discovery_service.ts(32,23): error TS2307: Cannot find module 
'node-uuid'.
services/discovery_service.ts(33,25): error TS2307: Cannot find module 
'xml2js'.
services/discovery_service.ts(103,33): error TS2304: Cannot find name 
'Buffer'.
services/media_service.ts(1,1): error TS6053: File 
'/home/pi/rpos-dev/rpos/typings/main.d.ts' not found.
services/media_service.ts(3,21): error TS2307: Cannot find module 'fs'.
services/media_service.ts(4,23): error TS2307: Cannot find module 'util'.
services/media_service.ts(7,22): error TS2307: Cannot find module 'url'.
services/media_service.ts(8,24): error TS2307: Cannot find module 'http'.
services/media_service.ts(19,26): error TS2304: Cannot find name 'require'.
[21:32:06] TypeScript: 50 semantic errors
[21:32:06] TypeScript: emit succeeded (with errors)
[21:32:06] Finished 'compile' after 14 s
[21:32:06] Finished 'default' after 14 s

BreeeZe, 2016-05-11:
Hi, thanks for your testing.
The typescript will indeed compile correctly without the definition files, they are used for checking the types during compiling and development but if they are missing typescript will just replace the type with "any".
The problem was, that I missed the "typings.install" call is asynchronous and returns a promise object..
So the compilation task was run before all typing files where present.
I fixed the gulp task, so now the 'compile' tasks is run when the typings are there instead of running in parallel.

So, I understand that you are directly working with the source on the Pi ? Are you also coding it on the Pi? or do you work on a separate machine? Which code editor are you using?

RogerHardiman, 2016-05-11:
Hi there

/> So, I understand that you are directly working with the source on the 
Pi ?//
//> //Are you also coding it on the Pi? or do you work on a separate 
machine? Which code editor are you using?/

I did most of the current changes on the Pi via a SSH session (putty) 
and 'vi' and the command line.
(I've been using Unix systems since 1989 and Linux since it 1992/1993 
and wanted to do things the old way.
(Normally I'm using VS, Eclipse or Xamarin Studio)

So edited, compiled and ran it on the Pi  (Pi 3)

Roger

BreeeZe, 2016-05-11:
That's pretty old school indeed :)
You do miss out on the advantage of typescript over regular javascript a bit though, having types and intellisense speeds up development quite a lot.
But i understand there is fun in doing things the hardcore old way :)

If you would like to use an editor, you should try out Vs Code, it's nice and light compared to the full VS and works really well with Typescript.

I'll merge the dev branch to the master.

closed 2016-05-11 by BreeeZe

RogerHardiman, 2016-05-11:
Hi
Will be using VS Code in the future.
3 weeks ago I have heard of NodeJS but had no idea what it was and I had 
never written any Javascript.
I had never heard of Typescript.

So it was good to learn things at the low level (command line tools) and 
will move on to VS Code next.

Roger