BreeeZe/rpos #16: GetScopes failed with error msg string (not xml/wsdl)

BlueT, 2016-05-16:
I'm testing rpos with node-onvif module https://github.com/agsh/onvif

the following functions are OK:
/onvif/device_service:
GetCapabilities
/onvif/media_service
GetProfiles
GetVideoSources

But this Failed:
/onvif/device_service
GetScopes

The content POST to rpos:
[POST-to-RPOS.txt](https://github.com/BreeeZe/rpos/files/266888/POST-to-RPOS.txt)

The content Responed from rpos:
[Response-from-RPOS.txt](https://github.com/BreeeZe/rpos/files/266890/Response-from-RPOS.txt)

BlueT, 2016-05-16:
[image]

RogerHardiman, 2016-05-16:
I have looked at the XML sent from agsh/onvif
The XML is not valid and there is a bug in the agsh/onvif code.

It would be better if RPOS returned an XML error message rather than the error msg string.

To help out I will do a fix for agsh/onvif and a Pull Request

RogerHardiman, 2016-05-16:
I have added a fix to agsh/onvif in this Pull Request [https://github.com/agsh/onvif/pull/37](https://github.com/agsh/onvif/pull/37)

BlueT, 2016-05-17:
update PR link https://github.com/agsh/onvif/pull/38

RogerHardiman, 2016-07-07:
This can be closed now.
The problem was in the agsh ONVIF library and not in RPOS.

BlueT, 2016-07-09:
Thanks for help. :-)

closed 2016-07-09 by BlueT