BreeeZe/rpos #22: The lastest version raspberry

Micka41, 2016-08-02:
I RPOS 0.1.0 and I run rpos.js all goes well
I tried the latest release of RPOS and impossible to launch (with ts-node?), I to a file not found error typing main.d.ts
How to launch the latest version of RPOS on Raspberry with PICAM?
Thank you.

RogerHardiman, 2016-08-02:
Hi
You have to type something like

npm build

Or run 'gulp' to convert .ts files into .js files

Roger

Micka41, 2016-08-06:
Hi
I have launch 
npm install
and then
ts-node rpos.ts

but i have this :

/usr/lib/node_modules/ts-node/src/index.ts:259
          throw new TSError(diagnosticList)
                ^
TSError: ⨯ Unable to compile TypeScript
rpos.ts (27,1): Cannot find name 'require'. (2304)
rpos.ts (29,23): Cannot find module 'http'. (2307)
rpos.ts (30,26): Cannot find module 'express'. (2307)
rpos.ts (38,13): Cannot find name 'require'. (2304)
rpos.ts (39,26): Cannot find name 'require'. (2304)
rpos.ts (2,1): File '/home/pi/rpos/typings/main.d.ts' not found. (6053)
    at getOutput (/usr/lib/node_modules/ts-node/src/index.ts:259:17)
    at /usr/lib/node_modules/ts-node/src/index.ts:268:16
    at Object.compile (/usr/lib/node_modules/ts-node/src/index.ts:404:17)
    at loader (/usr/lib/node_modules/ts-node/src/index.ts:290:33)
    at Object.require.extensions.(anonymous function) [as .ts] (/usr/lib/node_modules/ts-node/src/index.ts:307:14)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Function.Module.runMain (module.js:575:10)
    at Object.<anonymous> (/usr/lib/node_modules/ts-node/src/_bin.ts:179:12)

What does it means ?

Thanks

ts-node v1.2.2
node v6.3.1
npm 3.10.3
typesrcipt 1.8.10

RogerHardiman, 2016-08-07:
After your _npm install_ you need to run _gulp_

Just tested it on my Mac. checked out a fresh copy of the source tree, ran npm install and then ran gulp and it compiled fine.

Roger

RogerHardiman, 2016-08-07:
Just tried ts-node from my dev branch and it worked. Had to _npm install -g ts-node_ and _npm install -g typescript_ first but then it seemed to work.

RogerHardiman, 2017-05-14:
I've not heard from @Micka41 for 9 months so assume the problem was sorted.

closed 2017-05-14 by RogerHardiman