BreeeZe/rpos #23: Cannot start up rpos.js

mrlion420, 2016-08-04:
Hi,
I followed the instructions shown in text file. I used the following to start raspivid

root@raspberrypi:/home/pi/live/raspi# raspivid -o - -fps 25 -t 0 -b 800000 -n -w 1280 -h 720 | ./testRaspi

"h264" stream, from the file "stdin"
Play this stream using the URL "rtsp://192.168.1.123:8554/h264"

(We use port 80 for optional RTSP-over-HTTP tunneling.)

After this, I ran rpos.js. I have been losing my head over this for two days now. When i ran raspivid, it does not create /dev/video0 device as well.

root@raspberrypi:/home/pi/rpos-0.1.0# sudo node rpos.js
execSync('sudo cat /proc/cpuinfo')
Read IP address from awdl0 failed
Read IP address 192.168.1.123 from eth0
Manufacturer : Raspberry Pi
Model : 2 B
HardwareId : 
SerialNumber : 00000000f212401e
FirmwareVersion : 0.1.0
execSync('sudo modprobe bcm2835-v4l2')
Starting camera settings webserver on http://192.168.1.123:8081/
execSync('sudo v4l2-ctl --set-fmt-video=pixelformat=4')
Failed to open /dev/video0: No such file or directory
child_process.js:484
    throw err;
    ^

Error: Command failed: sudo v4l2-ctl --set-fmt-video=pixelformat=4
Failed to open /dev/video0: No such file or directory

    at checkExecSyncError (child_process.js:441:13)
    at Object.execSync (child_process.js:481:13)
    at Function.utils.execSync (/home/pi/rpos-0.1.0/lib/utils.js:94:66)
    at execV4l2 (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:173:22)
    at Object.SetPixelFormat (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:277:9)
    at Camera.setupCamera (/home/pi/rpos-0.1.0/lib/camera.js:129:27)
    at new Camera (/home/pi/rpos-0.1.0/lib/camera.js:44:14)
    at Object.<anonymous> (/home/pi/rpos-0.1.0/rpos.js:23:14)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)

RogerHardiman, 2016-08-04:
The command line you have used raspivid | ./testRaspi is not needed with RPOS. This command starts a RTSP server

RPOS has its own RTSP sever built in.

The /dev/video0 file is created when RPOS loads the kernel drivers for video capture.

mrlion420, 2016-08-05:
It was uv4l daemon. I was testing with it a while and it was using /dev/video0 when raspberry pi was booted. I killed the uv4l process and proceed on with the commands I've shown above. But now I cannot get it working on my Onvif device manager. It is saying "The envelope version of the incoming (Soap 11 (http://schemas.xmlsoap.org/soap/envelope/)) does not match that of the encoder (Soap 12 ....." 

I have check the wsdl files in rpos and is it using POST 1.1 ? If so, how I do add in POST  1.2 support?

mrlion420, 2016-08-08:
I solved it by changing the source file in rpos-0.1.0/node_modules/soap/lib/client.js. Changed `http://schemas.xmlsoap.org/soap/envelope/` to `http://schemas.xmlsoap.org/soap/envelope`.

closed 2016-08-08 by mrlion420

RogerHardiman, 2016-08-08:
RPOS 0.1.0 is quite old now. Version 0.2.0 was released on 10th May and this included some changes to the XML in node_modules/soap_client which I needed to make for ONVIF Device Manager.

I do not know if Breeeze has made an official release of 0.2.0. I just use git clone and get the source direct from Github.