BreeeZe/rpos #24: XML for Fault Codes

RogerHardiman, 2016-08-07:
Hi tried some of the unimplemented methods and verified them with the XML validity checker in SoapUI.

I think the fault responses are incorrect and should be like this (plus extra error codes for the future)
Even so, this is not 100% correct. It needs a lang='en' attribute on the Text field

    var SENDER_NOT_AUTHORIZED = {
      Fault: {
        Code: {
          Value: "soap:Sender",
          Subcode: {
            Value: "soap:NotAuthorized",
          }
        },
        Reason: {
          Text: "Sender not authorized"
        }
      }
    };

    var UNKNOWN_ACTION = {
      Fault: {
        Code: {
          Value: "soap:Sender",
          Subcode: {
            Value: "soap:UnknownAction",
          }
        },
        Reason: {
          Text: "Unknown Action"
        }
      }
    };

    var NOT_IMPLEMENTED = {
      Fault: {
        Code: {
          Value: "soap:Sender",
          Subcode: {
            Value: "soap:ActionNotSupported"
          }
        },
        Reason: {
          Text: "Optional Action Not Implemented"
        }
      }
    };

klukaspl, 2017-03-10:
Hi,

I have met similar problem with Text lang settings. I tried that:
 'Text xml:lang="en"' : "The service is not supported"

and in generated soap message I have xml:lang="en" present but on the both side of the section which is neither not correct.

Have you maybe found any solution for this?

BR,

Krzysztof

RogerHardiman, 2017-05-19:
Hi @klukaspl 
I had not looked at the error messages any more since writing this Problem Report 9 months ago, partly as I was not 100% sure of the format of the error other than Soap UI was saying the reply was not correct.