BreeeZe/rpos #25: no method execSync

IljaGrebel, 2016-10-27:
Hi, Im trying to start rpos with node using "node rpos.js", but I get an error message after few seconds

`/home/pi/rpos-0.1.0/lib/utils.js:94
            return utils.notPi() ? "" : require('child_process').execSync(cmd)
                                                                 ^
TypeError: Object #<Object> has no method 'execSync'
    at Function.utils.execSync (/home/pi/rpos-0.1.0/lib/utils.js:94:66)
    at Camera.loadDriver (/home/pi/rpos-0.1.0/lib/camera.js:126:15)
    at new Camera (/home/pi/rpos-0.1.0/lib/camera.js:41:14)
    at Object.<anonymous> (/home/pi/rpos-0.1.0/rpos.js:23:14)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
`

RogerHardiman, 2016-11-04:
What hardware, operating system version and node version are you using?

IljaGrebel, 2016-11-07:
Hi, Im using Raspberry Pi 2B+ with kernel version Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux, node version is v0.10.29.

RogerHardiman, 2016-11-08:
Hi
The execSync command does not exist in Node 0.10.x
It was added in Node 0.12.x
You need to upgrade NodeJS

Roger

RogerHardiman, 2016-12-01:
Did this fix your problem?

IljaGrebel, 2016-12-01:
Hi, yes I fix my problem :)

RogerHardiman, 2016-12-08:
Great. Could we close this PR?
Thanks

IljaGrebel, 2016-12-09:
Sure, sorry.

closed 2016-12-09 by IljaGrebel