BreeeZe/rpos #26: use as a gateway for RTSP streams?

syadnom, 2016-11-28:
How easily might it be to use rpos to pull an RTSP stream from another device instead of using video0?

There are a few cameras out there (a couple geovision, but mostly the ubiquiti UVC line) that doesn't support ONVIF despite being a decent h264 1080p camera.

I'd love to put a Pi in acting as a proxy and offering up those RTSP streams via an ONVIF interface.

RogerHardiman, 2016-12-01:
It is possible to do this.
There are two ways you can try
a) The ONVIF Command GetStreamURI returns the RTSP address for the video. So you could make it return the actual RTSP address of your camera. Your camera viewing software should then open the RTSP URL and get the video direct from the Pi

b) If your camera viewing software prevents you getting video from a different IP address to the ONVIF gateway then you will need to replace the RTSP server with one of your own.
Currently there are 2 RTSP servers - one was written by BreeZe and uses Live555 as the library. The other is written by mpromonet and also uses Live555.  I think live555 has a RTSP proxy but you could use any RTSP proxy for this. If you installed Mono on the Pi you could look at the RTSP proxy example that comes in the SharpRTSP project.

So it can be done

kristian, 2017-05-03:
Done, see [https://github.com/kristian/rpos-gateway](here). 👍

RogerHardiman, 2017-05-04:
Hi kristian
I've had a look at the changes. They are great.

I also added in partial gateway support in my rpos fork a few weeks ago [https://github.com/RogerHardiman/rpos/](https://github.com/RogerHardiman/rpos/commit/0af478d3f46115709721ce7795a0cc243b042e0d) but it was a quick trial and it still used the image size and gop information from the v4l layer (it was a quick demo for a project) so nice to see the extra config settings in your rpos-gateway to set these.

kristian, 2017-05-04:
Thanks Roger, appreciate it. Yes, I wanted to remove the dependencies to the camera include and the v4l layer. So I moved all settings to the rposConfig, so you can statically set-up the gateway as you please. Also I added some logic to the rpos-gateway.js (rpos.js) file, to allow hosting multiple gateways on the same Pi (by using the new --config parameter when starting node).

RogerHardiman, 2017-05-04:
Nice idea to have --config and to run several instances of rpos-gateway, with different  ports (e.g.port  8000, 8001, 8002) etc?

ONVIF does let you host lots of different streams via one ONVIF service. This capability is used by analogue camera encoders that have one IP address and one ONVIF service and have 16 analogue video inputs. ONVIF uses an input source token to identify which of the 16 the input you want stream information on.
This has the advantage that you are just running one discovery service for finding the device (the gateway) on the network but it does make it more complicated.

Anyway great piece of work and I may re-use some of the concepts in my RPOS with PTZ fork

kristian, 2017-05-04:
Hello Roger, I really appreciate your feedback, thanks. Yes, that's what I thought. Although, while testing the solution today, I noticed that it'll be required to add anohter `--no-discovery` parameter, as per the ONVIF specification the broadcast address & port are fix. So one host can only bind the port once and I don't want to introduce inter-process communication to find all runnign service ports to announce. So automated discovery works for one service per host only. A bummer.

I also noticed the possibility of ONVIF to host multiple streams. But, as you guessed, I wanted to keep things pretty straight forward, not altering the original rpos coding too much. Unless I receive a pull request, I don't think I'm going to add any multi-stream option. Supporting multiple rposConfig.json files was just so much more simple. :)

Thanks. Sure, I also saw your PTZ fork, which I think would make sense to pull into rpos honestly. On the other hand, I really think having a separate rpos-gateway fork makes sense, as it's another completely different use case.

RogerHardiman, 2017-05-06:
All my original enhancements to RPOS got included in the main project by Breeeze.
But the Pull Request for other improvements and other bug fixes is 9 months old now so I am not expecting anything new to go into RPOS at the current time.

RogerHardiman, 2017-05-14:
@kristian  Thanks for letting us know about the Gateway. Closing this PR down now but will add a line in the README about your project.

closed 2017-05-14 by RogerHardiman

dave-code-ruiz, 2020-12-16:
Hi, i am very interested in https://github.com/kristian/rpos-gateway, but when i try it, i see it is descontinuated, @kristian ¿it is posible you update code with last version of BreeeZe/rpos???

Thnks

RogerHardiman, 2020-12-16:
The main RPOS project includes gateway functionality. There is an option in the json Config file to specify the RTSP source which could be an RTSP address of another far-away device.
Note that some VMS systems may not like this and may assume the IP Address of the ONVIF service is the same IP address as the RTSP server. But the ones I tried were happy with it.

VCTGomes, 2023-07-31:
> json

Incredible, Roger! It's working now and my camera is being exported to onvif. I'm using rposConfig.sample-proxy.json, but I'd like to know if is it possible to export the secondary streaming too?

My NVR is able to catch mainly streaming, but the image is black when using secundary streaming like on data mobile and more.

Is it possible to set up motion sensor using some web hocks too?