BreeeZe/rpos #28: RPi newbie struggling to get this to work

Himbeer-Torte, 2016-12-07:
I'm struggling to get this to work. I've worked through the read.me instructions and installed everything correctly (I think). I get some error messages when I run the sudo node rpos.js command.
I'm using an RPi3 with a RPi Camera module v1.3.

RogerHardiman, 2017-02-08:
My own build process is
1) get the source from github
2) 'cd' into the rpos directory
3) run 'npm install' to load the dependencies. On a RPi this takes a very long time.
4) run 'gulp' to compile the TypeScipt into JavaScript
5) Edit the rposConfig.json file
6) run rpos with 'sudo node rpos.js'

BlueT, 2017-02-20:
@Himbeer-Torte what error message did you get?

iotoi8cn, 2017-03-01:
can you create the release for this current code? thank you.

cnruby, 2017-03-01:
can you create the release for this current code? thank you.

jjanderson, 2017-03-19:
Hi, 

I followed the steps, however when I ran the start command received this error:

Using IP address from config: [removed]
Manufacturer : Raspberry Pi
Model : 2 B
HardwareId :
SerialNumber : 0000000041bcaa83
FirmwareVersion : 0.1.0
Starting camera settings webserver on http://192.168.1.18:8081/
child_process.js:484
    throw err;
    ^

Error: Command failed: sudo v4l2-ctl --set-fmt-video=pixelformat=4

    at checkExecSyncError (child_process.js:441:13)
    at Object.execSync (child_process.js:481:13)
    at Function.utils.execSync (/home/pi/rpos-0.1.0/lib/utils.js:94:66)
    at execV4l2 (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:173:22)
    at Object.SetPixelFormat (/home/pi/rpos-0.1.0/lib/v4l2ctl.js:277:9)
    at Camera.setupCamera (/home/pi/rpos-0.1.0/lib/camera.js:129:27)
    at new Camera (/home/pi/rpos-0.1.0/lib/camera.js:44:14)
    at Object.<anonymous> (/home/pi/rpos-0.1.0/rpos.js:23:14)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)

Any ideas

RogerHardiman, 2017-05-14:
@jjanderson   when you had this error, was the Pi camera installed and working and can you tell me the version of Node you were using?

jerseyj, 2017-05-19:
I had the exact same issue as jjanderson! I installed the latest version of pos and followed the directions exactly. I also since I got the error code made sure to be using the latest node.js as well. It didn't help.

The directory name is pos-0.1.0 so I'm guessing that's the version ?

RogerHardiman, 2017-05-19:
pos-0.1.0 is a snapshot of the source from 2015. Very old and out of date.

In step 1 at the top of this post it said 'get the source from github' and I was not very clear there. What I meant was to get the source with 'git clone' so you had the very latest source code. So apologies for not making that clear.

Anyway to help out I've made a 2.0.0 source release.

jerseyj, 2017-05-19:
Thanks for the update...i'm a total newbie at all of your his, how would i get and install 2.0 on my Pi ? would a "get update " work ?

- "Defeat lasts one day, giving up lasts a lifetime."

RogerHardiman, 2017-05-29:
Yiu can down load it from the Releases tab

RogerHardiman, 2017-06-15:
Have not had any updates so closing this for now.

closed 2017-06-15 by RogerHardiman

jjanderson, 2017-07-19:
Hi, Sorry for the delay in getting back to you..

I have now followed the instructions and the camera is being picked up by Zoneminder via ONVIF. However in ZM I have a blue screen.

I am using a Genius 120-degree Ultra Wide Angle Full HD Conference Webcam(WideCam F100) on a Raspberry PI 3.

I am however able to browse to port 8081 on the pi and view the settings..

Sorry for the pestering noobness of my queries, but would love to get this working,

RogerHardiman, 2017-07-20:
The Genius camera is a USB camera and so will not work.

RPOS uses the official Raspberry Pi camera which does not use USB.
So it is great that you have got ZoneMinder to connect to RPOS, but if you don't have an official Raspberry Pi camera then you will not get any video.

It would be nice in the future to support USB cameras but I don't have time for that right now.

If anyone reading this wants to help, the github user called mpromonet has a userspace driver that can capture from USB, compress with H264 and make it available as a /dev/video source which RPOS could use. But I have never tried it and don't have time to.

So currently (July 2017) RPOS can only work with the official Raspberry Pi camera.

jjanderson, 2017-07-20:
Ok, thanks for getting back to me.

I will get a pi cam which I intended to do anyway

I can update the documentation for n github if you like, but that is about the extent of my prowess.. I can help "dumb it down" as my initial search in google was for "covert pi with USB camera to ip camera" and when I landed on the page, it was not clear that rpos did not work with USB.. not a problem, it may just weed out more silly questions from noobs such as myself... otherwise, thanks for an awesome app! Cant wait to get my picam now! 😀

pacman8521, 2019-01-07:
RogerHardiman if you are out there I could use some help.  I've gone through the rpos install twice and am having a few minor problems.  I feel as though all of the components are installed correctly but am unable to view the stream.  I'm stuck.  My end goal is to set up a RTSP camera on raspberry and link it into my Geovision Recording server as a RTSP feed across my network.  I am unable to browse the feed in ONVIF Device Manager.

RogerHardiman, 2019-01-07:
recent addition this thread is a duplicate of #54. remaining closed