BreeeZe/rpos #30: Problem setting bitrate through web interface

rungkutta, 2017-02-04:
While using rpos, I noticed that when setting `video_bitrate` through the web interface, the value is not preserved.  The value always reads '0' in the `v4l2ctl.json` settings file.  I tracked this down to what I think is a RegExp bug in `v4l2ctl.ts`

In the `ReadControls()` function, we take the control name and append the variable `regexPart` to create a RegExp that is matched against the output of `v4l2-ctl -l` command.

The problem shows when two different controls start the same way: in this case `video_bitrate_mode` and `video_bitrate`.  The regex will match `video_bitrate_mode` for both controls.  So, when you attempt to set `video_bitrate`, it takes affect and even gets written to file correctly.  But anytime the file is read, the value of `video_bitrate_mode` will be captured for both controls.

To fix, you could change

    var regexPart = ".*value=([0-9]*)";

to

    var regexPart = "\\s.*value=([0-9]*)";

This will require whitespace after the control name, which will limit the RegExp to only match the correct control.

RogerHardiman, 2017-02-06:
Have a tiny suggestion on the Regex. I think some parameters can go direct into the Colon eg
`something: value-42`

So how abut a regex that is
[\s:].*value=([0-9]*)

RogerHardiman, 2017-02-06:
Thanks for the bug report.
I had seen this problem but never had the time to investigate so thank you for the fix.

Will we always have a whitespace after the name in v4l2-ctl?
Is that part of the formatting?

rungkutta, 2017-02-06:
It's a good question, I'm not sure.  I also considered \b for a word break, which would catch both a whitespace or a colon.  However, this wouldn't allow for names with a hyphen (-) in them if those exist (it would allow underscores).

Just remember to escape the slash in the string, or you'll really confuse yourself.

RogerHardiman, 2017-02-08:
I am back in the office now and checked my RPi.
The original regex is fine.
I have a fork of rpos (with extra things like PTZ support and Aux/Relay support) and will add it into my fork too.

rungkutta, 2017-02-09:
For the record, we should note that the sense of auto_exposure is backwards in these settings.  A value of 0, or unchecked, means auto exposure is on.  A value of 1 or checked means manual exposure.  Not sure the easiest way to note this in the interface, but I suppose you could change it to a menu/dropdown type where "on" = 0 and "off" = 1.

RogerHardiman, 2017-05-14:
This bug has been fixed in RPOS now. (I've pulled in the Pull Request from my fork).
Thanks again for the bug report

closed 2017-05-14 by RogerHardiman