BreeeZe/rpos #31: Problem with discovery probe message for tds:Device

klukaspl, 2017-03-08:
Hi,

I am trying to change the part of discovery_service.ts to respond not as NetworkVideoTransmitter but as Device. In if statement of parseString function I changed NetworkVideoTransmitter to Device, as well as in SOAP message I replaced <d:Types>dn:NetworkVideoTransmitter</d:Types> to be <d:Types>tds:Device</d:Types>.

I am testing that with Onvif Test Tool 17 and seems this tool is sending first Device and then NetworkVideoTransmitter probe. While I can see that the message with my changes is replied - I am unable to discover the device with the tool. With NetworkVideoTransmitter it is working. 

Can anyone give some hint how to fix that ?

Regards,
Krzysztof

RogerHardiman, 2017-05-16:
Hi,
Thanks for the information on Test Tool 17 Probing.
It may be a surprise, but I don't have ONVIF Test Tool.
Can you send over some logs of the Discovery Messages Test Tool Emits?

Thanks

klukaspl, 2017-05-16:
Hi,
I have changed the reply to this:
   let reply = `<?xml version="1.0" encoding="UTF-8"?>
          <SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl">
            <SOAP-ENV:Header>
              <wsa:MessageID>uuid:${uuid.v1()}</wsa:MessageID>
              <wsa:RelatesTo>${probe_uuid}</wsa:RelatesTo>
              <wsa:To SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:To>
              <wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsa:Action>
              <d:AppSequence SOAP-ENV:mustUnderstand="true" MessageNumber="34" InstanceId="1488957280"></d:AppSequence>
            </SOAP-ENV:Header>
            <SOAP-ENV:Body>
              <d:ProbeMatches>
              <d:ProbeMatch>
                <wsa:EndpointReference>
                  <wsa:Address>urn:uuid:5581ad80-95b0-11e0-b883-accc8e0c06bb</wsa:Address>
                </wsa:EndpointReference>
                <d:Types>tds:Device</d:Types>
                  <d:Scopes>
                    onvif://www.onvif.org/location/
                  </d:Scopes>
                  <d:XAddrs>http://${utils.getIpAddress()}:${this.config.ServicePort}/onvif/device_service</d:XAddrs>
                  <d:MetadataVersion>1</d:MetadataVersion>
                  </d:ProbeMatch>
                  </d:ProbeMatches>
              </SOAP-ENV:Body>
              </SOAP-ENV:Envelope>`;
and now it is working and present itself as Device.

BR,

Krzysztof

RogerHardiman, 2017-05-17:
Thanks for the fix.
What I'll do is return this new XML if the scope is Device, and the old XML if the scope is NetworkVideoTransmitter.
If you wanted to fork, implement and raise a PullRequest that would be great.

Thanks
Roger

klukaspl, 2017-05-18:
I am trying to implement that and as well Hello and Bye messages to have Reboot and some network related methods, working as it is specified by onvif. But I have problem probably with asynchronous calls of the udp sockets calls. While I have the Hello working correctly, the Bye message is not sent before I am closing discovery socket (or just reboot). Any idea how to cope with that?

RogerHardiman, 2017-05-18:
That would  be great if you could implement Hello and Bye.
For the Bye we probably need a small delay after sending the UDP packet and calling the reboot.
I did check to see if we could use shutdown -r TIME in the reboot code
  ```utils.execSync("sudo reboot")```
becomes
  ```utils.execSync("sudo shutdown -r now")```
This may give enough delay. But we may want to use it anyway as it does a graceful shutdown.

But the best bet is probably to put in a timer
So change
  ```utils.execSync("sudo shutdown -r now")```
to something like this
  ```setTimeout(function(){utils.execSync("sudo shutdown -r now")},3000);// schedule reboot in 3 seconds```

klukaspl, 2017-05-22:
Hi
thanks - I will check that.

I have one another struggle here. The discovery we have here is listening on multicast address. So it fails when I try perform unicast (which is used after reboot for example). Any idea how to cope with that?
I was thinking to run one another socket but no result.

BR
Krzysztof

RogerHardiman, 2017-05-22:
Hi @klukaspl 
I would start a 2nd Discovery Service on the unicast address of th RPOS device.
I will add that I've never seen unicast discovery messages in use before. I think unicast is used with Discovery Proxies but I've only ever seen the Multicast discovery system in use.
What program was using the unicast discovery protocol? (was it from Test Tool or a VMS?)

klukaspl, 2017-05-23:
Hi @RogerHardiman 
There is the test : SEARCH USING UNICAST PROBE MESSAGE. which I am currently not able to pass. It seems that there is similar step when testing reboot. But I think you are right that there could be applied only to Discovery Proxy.

RogerHardiman, 2017-06-15:
Did you have any changes or improvements to add to rpos?

closed 2018-12-31 by RogerHardiman