BreeeZe/rpos #36: error TS2307: Cannot find module 'ip'.

acshun, 2017-07-10:
I was following the steps and when I ran ./node_modules/gulp/bin/gulp.js I receive the following output:
[17:12:24] Using gulpfile ~/rpos/gulpfile.js
[17:12:24] Starting 'default'...
[17:12:24] Starting 'typings'...
[17:12:26] Finished 'typings' after 2.09 s
[17:12:26] Starting 'compile'...
services/device_service.ts(10,21): error TS2307: Cannot find module 'ip'.
[17:12:38] TypeScript: 1 semantic error
[17:12:38] TypeScript: emit succeeded (with errors)
[17:12:38] Finished 'compile' after 12 s
[17:12:38] Finished 'default' after 14 s

Any ideas on how to resolve this error?

HSorensen, 2017-07-16:
Just a me too. 
I did 
'''npm install ip'''
But the problem persisted.

RogerHardiman, 2017-07-17:
Fixed in latest commit

The 'ip' module was not defined in the Typescript typings file

closed 2017-07-17 by RogerHardiman

shmrymbd, 2018-08-26:
hi roger, i still have the problem on latest version. pls advice how to