BreeeZe/rpos #37: ONVIF Manager

MarkReese, 2017-08-15:
It works, but ONVIF manager does not discover it on the network?

RogerHardiman, 2017-08-19:
Can you tell me more about your setup.
Is this ONVIF Device Manager on Windows or some other ONVIF Manager.
Are you running RPOS from the latest source in Github or from a release build
What OS are you running RPOS on
Which version of NodeJS are you using for RPOS

Thanks

MarkReese, 2017-08-22:
Hello,

                Thanks for getting back to me, I used the info from here https://github.com/BreeeZe/rpos and was never able to get it to work properly.  I was able to get https://github.com/kristian/rpos-gateway this fork to work, but discovery still doent work. Then I noticed they added no discovery as an option, don’t kmow where or how you set that.

 

Raspberry PI 2B

ODM is on windows 7 Pro

Raspberry PI is Running Raspbian

NodeJS on your GitHub I think is v6, on the fork it is v7