BreeeZe/rpos #38: Installation on Stretch

robmarkcole, 2017-09-25:
On a pi zero W with Raspbian GNU/Linux 9 stretch, nodejs version 4.8.2, I receive the following errors when following the installation instructions (note that previous steps completed OK):

```
pi@raspberrypi:~/onvif/rpos $ sudo npm install
npm WARN deprecated typings@0.8.1: Typings is deprecated in favor of NPM @types -- see README for more information
npm WARN deprecated node-uuid@1.4.7: Use uuid module instead
npm ERR! 404 Not Found
npm ERR! 404
npm ERR! 404 'gulp-sourcemaps/map-sources' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it
npm ERR! 404 It was specified as a dependency of 'gulp-sourcemaps'
npm ERR! 404
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, or http url, or git url.

npm ERR! System Linux 4.9.41+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/onvif/rpos
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code E404
npm ERR! fetch failed https://registry.npmjs.org/update-notifier/-/update-notifier-0.6.3.tgz
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.
-
> serialport@4.0.7 install /home/pi/onvif/rpos/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@4.8.2 (node-v46 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/onvif/rpos/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/onvif/rpos/node_modules/serialport/build'
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/onvif/rpos/npm-debug.log
npm ERR! not ok code 0

pi@raspberrypi:~/onvif/rpos $ ./node_modules/gulp/bin/gulp.js
module.js:327
    throw err;
    ^

Error: Cannot find module 'gulp-sourcemaps'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/pi/onvif/rpos/gulpfile.js:8:18)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
```

rpelissi, 2017-12-02:
I have the same issue on strech also.

RogerHardiman, 2017-12-02:
I read something about this being caused by old put of date versions of npm.

Note that in the installatuon tutorial I recommend node v6 and the version of npm that comes with it.

closed 2018-12-31 by RogerHardiman