BreeeZe/rpos #39: ERR_SERVER_ALREADY_LISTEN on running rpos.js

jov58, 2017-12-05:
I'm trying to get this to work on a Raspberry Pi Zero W. But when I run the final command (sudo node rpos.js) I get this error. Does anybody have an idea how to resolve this?

Read IP address 192.168.1.130 from wlan0
Manufacturer : Raspberry Pi
Model : Zero W
HardwareId :
SerialNumber : 000000002d926624
FirmwareVersion : 2.0.0
Starting camera settings webserver on http://192.168.1.130:8081/
Binding DeviceService to http://192.168.1.130:8081/onvif/device_service
Binding MediaService to http://192.168.1.130:8081/onvif/media_service
Uncaught Exception... : Error [ERR_SERVER_ALREADY_LISTEN]: Listen method has been called more than once without closing.
    at Server.listen (net.js:1446:11)
    at MediaService.SoapService.start (/home/pi/rpos/lib/SoapService.js:26:24)
    at Object.<anonymous> (/home/pi/rpos/rpos.js:32:15)
    at Module._compile (module.js:641:30)
    at Object.Module._extensions..js (module.js:652:10)
    at Module.load (module.js:560:32)
    at tryModuleLoad (module.js:503:12)
    at Function.Module._load (module.js:495:3)
    at Function.Module.runMain (module.js:682:10)
    at startup (bootstrap_node.js:191:16)

RogerHardiman, 2017-12-05:
usually that means you already have a Web Server listening on port 8081 already

You can change the port that RPOS listens on in the rposConfig.json file

jov58, 2017-12-05:
That would be odd, because I used a brand new SD card, newly flashed with Raspbian Jessie. In that case, only the installation could have caused this? Speaking of installation, pls see my other post.

Stuip, 2018-04-07:
The problem solved ?

jov58, 2018-04-08:
Nope. Stopped trying, busy doing other things.

closed 2018-04-08 by jov58

RogerHardiman, 2018-12-31:
Hi @jov58 and @Stuip 
I finally tracked down the cause of ERR_SERVER_ALREADY_LISTEN with the RPOS project.
There were some network changes in Node JS version 9 and version 10 which broke RPOS.

I'd only ever used RPOS with Node v6 and Node v8 so I never saw the problem.

Anyway it has just been fixed in the RPOS source code.