BreeeZe/rpos #40: Installation instructions?

jov58, 2017-12-05:
I found this project because I want to have a few Raspberries (preferably Zero W's) with Pi camera's around my house, connected to Surveillance Station on my SynologyDS1511+. I am fairly new to Linux.

But the installation instructions confuse me. I find something [here](https://github.com/BreeeZe/rpos), but also [here](http://breeeze.github.io/rpos/). Do I have to combine these or use only one of them?

jov58, 2017-12-06:
OK, I came to a working system on a Raspberry Pi 3 running a fresh Jessie (not Lite) image:

1. Install <b>live555</b> with the instructions on [this](https://breeeze.github.io/rpos/) page.
2. Install <b>nodejs</b> with the instructions on that same page.
3. Follow the instructions on [this](https://github.com/BreeeZe/rpos) page, starting at the third line from step 2, also at <I>sudo apt-get install liblivemedia-dev</I>.
4. Add the Raspberry as a ONVIF camera with port 8081.

I also noticed that you have to cd to the rpos directory before running rpos.js. Something like <I>sudo node /home/pi/rpos/rpos.js</I> while in another directory than rpos, does not work. I added these two lines to /etc/rc.local to have rpos started at boot:
```
   cd /home/pi/rpos
   sudo node rpos.js &
```
and it works :) ! I can restart from Surveillance Station.

BTW, I always get this error at the moment Surveillance Station contacts the Raspberry:

`rtspServer: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
`

Output on the Raspberry Pi 3 when connected to Surveillance Station:

```
Read IP address 192.168.16.138 from wlan0
Manufacturer : Raspberry Pi
Model : 3
HardwareId :
SerialNumber : 0000000012345678
FirmwareVersion : 2.0.0
Starting camera settings webserver on http://192.168.16.138:8081/
Binding DeviceService to http://192.168.16.138:8081/onvif/device_service
Binding MediaService to http://192.168.16.138:8081/onvif/media_service
Binding PTZService to http://192.168.16.138:8081/onvif/ptz_service
discovery_service started
Starting Live555 rtsp server
media_service started
device_service started
ptz_service started

rtspServer: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!
```
=============================

On a Raspberry Pi Zero W it's almost the same, albeit rather slow. I used a fresh Jessie Lite image here. And right before running the git command during point 3 above, you have to do this:

```
sudo apt-get update && sudo apt-get install -y git
```
Surveillance Station accepts the Pi Zero as an ONVIF camera, but then it fails: it can't activate it (see the error below). In the properties of the camera (on Surveillance Station) I've selected lowest bandwidth everywhere and deselected any fancing looking stuff, in case it might be some timing problem - it didn't help. I didn't yet try to change any parameter on the Pi Zero.

Output on the Raspberry Pi Zero W when connected to Surveillance Station:

```
Read IP address 192.168.16.130 from wlan0
Manufacturer : Raspberry Pi
Model : Zero W
HardwareId :
SerialNumber : 0000000087654321
FirmwareVersion : 2.0.0
Starting camera settings webserver on http://192.168.16.130:8088/
Binding DeviceService to http://192.168.16.130:8088/onvif/device_service
Binding MediaService to http://192.168.16.130:8088/onvif/media_service
Binding PTZService to http://192.168.16.130:8088/onvif/ptz_service
discovery_service started
device_service started
Starting Live555 rtsp server
media_service started
ptz_service started
rtspServer: VIDIOC_STREAMON: Invalid argument
VIDIOC_DQBUF: Invalid argument
```

Comments on what I did are welcome! As are ideas to get it working on the Pi Zero W ...

RogerHardiman, 2018-12-30:
Hi
Sorry for the 12 month delay in getting back to you on this.
I don't own a Pi Zero W so I cannot try this myself.
What I can see from your error is that the RTSP server had problems talking to the v4l2 driver and /dev/video0

rtspServer: VIDIOC_STREAMON: Invalid argument
VIDIOC_DQBUF: Invalid argument

So maybe there was an issue with the camera or an issue with the bcm2835-v4l2 driver.

I wondered if you had given it another try in the last year and if there was any new updates?

jacobalberty, 2018-12-31:
I have it working with the bcm2835-v4l2 driver and the mpromonet/v4l2rtspserver on a pi zero w, only caveat I found is I'm not sure 128mb is enough gpu ram, I upped it to 256mb

RogerHardiman, 2018-12-31:
good to hear it is working now on the Pi Zero W.
Maybe the original poster had an old linux/bcm driver so I'll close this Issue report down.

closed 2018-12-31 by RogerHardiman