BreeeZe/rpos #41: rtspServer vs v4l2rtspserver / DVR multi-view screen issues

programwitch, 2017-12-22:
We recently got RPOS up and running on both a Raspberry Pi Zero W and a Pi 3.

When using rtspServer ("RTSPServer" : 1 in rposConfig.json) we would get
errors including: "rtspServer: StreamParser::afterGettingBytes() warning: read 4294967295 bytes: expected no more than 15000"

We then switched over to using v4l2rtspserver ("RTSPServer" : 2 in rposConfig.json) and started getting a nice stable image without any errors.

But there is a new issue on the other end of things at the Security Camera DVR, we cannot see the Raspberry Pi's video feed in any multi view screen (where you can see 4 or more cameras at once). We can only see the video feed in single view (just viewing our Raspberry Pi camera). (Note: Tested on two different models of DVR with the same results.)

Has anyone had this issue before? Is there another setting needed for v4l2rtspserver?

RogerHardiman, 2017-12-22:
I have seen some CCTV software that requires IP cameras with 2 streams (eg one at 1920x1080 and obe at 640x480) and when the system show 4 or more images it looks for a profile that is close to 640x480 pixels.
At the current time RPOS only outputs 1 stream so there is no 2nd stream.

This may be your problem. If this is the problem then he only fix I have is to run RPOS at a lower resolution.

There is probably some clevery way to get two different streams from the GPU but I don't think we can do that via the v4l2 interface.

esiexata, 2018-02-14:
if possible to create 2 identical streams to work whith  DVRs systens? not generate 2 streans, only clone to first stream.

RogerHardiman, 2018-12-28:
Yes it would be possible. You would need to edit the code for GetProfiles and create a 2nd profile  and then return the same url

closed 2018-12-31 by RogerHardiman