BreeeZe/rpos #43: sudo node rpos.js

Stuip, 2018-04-13:
`Read IP address 192.168.1.196 from eth0
Manufacturer : Raspberry Pi
Model : 2 B
HardwareId :
SerialNumber : 0000000099774564
FirmwareVersion : 2.0.0
Starting camera settings webserver on http://192.168.1.196:8081/
Binding DeviceService to http://192.168.1.196:8081/onvif/device_service
Binding MediaService to http://192.168.1.196:8081/onvif/media_service
Uncaught Exception... : Error [ERR_SERVER_ALREADY_LISTEN]: Listen method has bee                   n called more than once without closing.
    at Server.listen (net.js:1425:11)
    at MediaService.SoapService.start (/home/pi/rpos/lib/SoapService.js:26:24)
    at Object.<anonymous> (/home/pi/rpos/rpos.js:32:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)`

how to solved the problem?

edbir1, 2018-08-23:
Is there any movement on this error?  I am receiving the same error.  And it does it on  media_service.start(), ptz_service.start() and imaging_service.start()

Starting camera settings webserver on http://192.168.5.181:18082/
Binding DeviceService to http://192.168.5.181:18082/onvif/device_service
Binding ImagingService to http://192.168.5.181:18082/onvif/imaging_service
Uncaught Exception... : Error [ERR_SERVER_ALREADY_LISTEN]: Listen method has been called more than once without closing.
    at Server.listen (net.js:1425:11)
    at ImagingService.SoapService.start (/home/pi/rpos/lib/SoapService.js:26:24)
    at Object.<anonymous> (/home/pi/rpos/rpos.js:36:17)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)
pi@raspberrypiADSB:~/rpos $ node -v
v9.11.2
pi@raspberrypiADSB:~/rpos $ npm -v
5.6.0

edbir1, 2018-08-23:
it appears that rpos does not like Node v9.  Node v8.11.1 will work.

RogerHardiman, 2018-12-30:
Thanks for this. Have updated Readme to say it works with v6 and v8

closed 2018-12-30 by RogerHardiman

RogerHardiman, 2018-12-31:
Good news. 
I have upgrade one of my machines to Node V10 and I could re-create the  [ERR_SERVER_ALREADY_LISTEN] issue you had reported.
The problem was caused by a change to the http and network libraries in NodeJS so it threw an error/exception rather than just ignoring a problem.

I've now fixed the problem in the source code of RPOS.

Thanks for the Problem Report