BreeeZe/rpos #44: gulp compile fails; node 8.x

peebles, 2018-08-05:
```sh
pi@raspberrypi:~/rpos $ npx gulp
[17:23:05] Using gulpfile ~/rpos/gulpfile.js
[17:23:05] Starting 'default'...
[17:23:05] Starting 'typings'...
[17:23:11] Finished 'typings' after 5.89 s
[17:23:11] Starting 'compile'...
/home/pi/rpos/typings/main/ambient/ip/index.d.ts(8,1): error TS1084: Invalid 'reference' directive syntax.
[17:23:25] TypeScript: 1 syntax error
[17:23:25] TypeScript: emit succeeded (with errors)
[17:23:25] Finished 'compile' after 15 s
[17:23:25] Finished 'default' after 20 s
pi@raspberrypi:~/rpos $ node --version
v8.11.1
pi@raspberrypi:~/rpos $ npm --version
6.1.0
```

RogerHardiman, 2018-12-30:
Thanks for the bug report.
If you want to try and fix it and do a Pull Request that would be great.

Thanks

RogerHardiman, 2018-12-30:
Good news... I've fixed the issue (and a 2nd issue it was masking). The changes are in git.

I also learned something. I had not seen 'npx' before. That is really handy.

Thanks for the Issue Report

closed 2018-12-30 by RogerHardiman