BreeeZe/rpos #46: Focus commands are not transmitted

helgboy, 2018-10-11:
When I send the zoom in/out onvif commands to the RPI, the zoom command is transmitted to the RS232 port. 
I get on the rs232 following code (Pelco D):  
FF 01 00 20 00 00 21
FF 01 00 40 00 00 41 
When I send the focus near/far onvif commands to the RPI, the focus commands are not transmitted to the RS232 port. 
But I get the STOP command when I send focus near/far. See code (Pelco D):
FF 01 00 00 00 00 01  - it is not right !
I should to get:
FF 01 00 80 00 00 81 
FF 01 01 00 00 00 01

RogerHardiman, 2018-10-21:
The problem is because the Focus Commands are not handled in lib/PTZDriver.ts
Unhandled PTZ/Imaging Command Received: focus Value:0.600000024
Unhandled PTZ/Imaging Command Received: focusstop
Unhandled PTZ/Imaging Command Received: focus Value:-0.600000024
Unhandled PTZ/Imaging Command Received: focusstop

I will try and implement these

RogerHardiman, 2018-10-21:
This should be fixed now.

closed 2018-10-27 by RogerHardiman