BreeeZe/rpos #48: npm install

arduiacob, 2018-11-11:
pi@raspberrypi:~/rpos $ sudo npm install
(node:8782) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN package.json soap@0.8.0 No license field.
npm WARN package.json node-pelcod@0.3.0 No repository field.
npm WARN package.json node-uuid@1.4.8 No license field.
npm WARN package.json pan-tilt-hat@1.1.0 No repository field.
npm WARN package.json tenx-usb-missile-launcher-driver@0.2.0 No repository field.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.71-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/rpos
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/rpos/npm-debug.log
npm ERR! not ok code 0

gustavoguimas, 2018-11-12:
I solve it by installing npm using this guide
https://www.w3schools.com/nodejs/nodejs_raspberrypi.asp

hope it helps

jacobalberty, 2018-12-30:
@gustavoguimas  only problem with that is I'm trying to use a raspi zero w and nodesource doesn't have armv6 debs. If I install node 10 armv6 tarball from nodejs.org I can run npm install fine but then I can't execute "node rpos.js" If I do npm install on node 10 then remove node 10 and install node 8 from the distro packages I can t hen run "node rpos.js", this is kind of hacky though.

jacobalberty, 2018-12-30:
Ok, I think I got it licked:
```
apt-get update
apt-get install nodejs npm`
npm install -g npm@latest
/usr/local/bin/npm install
```
I'm not sure how to replace the deb installed npm binary with the updated npm binary cleanly but this route will generate a working install of rpos using the debian packages. It's still a little hacky but it doesn't require installing an entire copy of nodejs that you're just going to remove later

An alternative you might consider trying is the following:
```
apt-get update
apt-get install nodejs
curl -L https://npmjs.org/install.sh | sh
npm install
```

I haven't tested this route yet but in theory it will let you install just nodejs from debian and then install npm separately so you only ever install one copy of npm. Alas installing rpos on armv6 is going to be a little hacky since theres no nodesource packages available.

RogerHardiman, 2018-12-30:
I updated the readme file earlier today with some new instructions that included upgrading npm so the posts today on this Issue are great.  If you think it is worth updating the Readme then let me know.

jacobalberty, 2018-12-30:
Ah I see now you did say to update npm. I think once you've run npm once in that session bash will cache the location so it doesn't have to search the full PATH. a logout and login after updating npm may be another solution to getting it to recognize the new binary.

RogerHardiman, 2018-12-31:
I made a change to rpos today that made it run with Node 10, so no need to downgrade to Node 8.

I