BreeeZe/rpos #50: ERR_SERVER_ALREADY_LISTEN

dony71, 2018-12-27:
Read IP address 10.168.1.40 from eth0
Manufacturer : Banana Pi
Model : M2+
HardwareId :
SerialNumber : 02c000812232c635
FirmwareVersion : 0.2.1
Binding DeviceService to http://10.168.1.40:8880/onvif/device_service
Binding MediaService to http://10.168.1.40:8880/onvif/media_service
net.js:1372
    throw new ERR_SERVER_ALREADY_LISTEN();
    ^

Error [ERR_SERVER_ALREADY_LISTEN]: Listen method has been called more than once without closing.
    at Server.listen (net.js:1372:11)
    at MediaService.SoapService.start (/root/rpos-gateway-0.2.1/lib/SoapService.js:26:24)
    at Object.<anonymous> (/root/rpos-gateway-0.2.1/rpos-gateway.js:42:15)
    at Module._compile (internal/modules/cjs/loader.js:688:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:699:10)
    at Module.load (internal/modules/cjs/loader.js:598:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:537:12)
    at Function.Module._load (internal/modules/cjs/loader.js:529:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:741:12)
    at startup (internal/bootstrap/node.js:285:19)

RogerHardiman, 2018-12-28:
Thanks for the issue report.
I have not used a Banana Pi before but it would be fine as RPOS will work on a Mac and Windows.

What version of Node are you running?

RogerHardiman, 2018-12-28:
Just realised this is a bug report for rpos-gateway.  That is a different project, forked from RPOS. You will need to contact the rpos-gateway team.

If you can recreate the issue with RPOS then please create a new Problem Report and I can take a look.

closed 2018-12-28 by RogerHardiman

RogerHardiman, 2018-12-31:
Hi @dony71 
Quick update - I was able to re-create the issue in the RPOS project when running on Node JS version 10. It may have occurred in Node v9 too.

Anyway I've fixed RPOS so it no longer triggers ERR_SERVER_ALREADY_LISTEN when running on Node version 10.
You are using roos-gateway so you will need to get the roos-gateway author to make the same changes.