BreeeZe/rpos #53: Device not detect correctly from Milestone

prometrix, 2019-01-03:
Im not able to add the device to Milestone (2018 R3) i get pass the first step to add new hardware and it finds the correct hardware model text. But the next step fails with error message:

--
It was not possible to connect to or receive information from the hardware '192.168.123.110'.
Verify that the following is correct: address, port, user name, password and hardware driver.
Alternatively use auto-detection to find the hardware.
--

Im using the built in rtsp and the pi camera. I can access the RTSP url from VLC. 

Any ideas?

This is the last debug output i get:
--
DeviceService - Calltype : info, Data : Handling POST on /onvif/device_service
DeviceService - Calltype : received, Data : <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" 
    xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" 
    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
    xmlns:xsd="http://www.w3.org/2001/XMLSchema" 
    xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" 
    xmlns:ds="http://www.w3.org/2000/09/xmldsig#" 
    xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" 
    xmlns:tt="http://www.onvif.org/ver10/schema" 
    xmlns:tds="http://www.onvif.org/ver10/device/wsdl" 
    xmlns:wsa5="http://www.w3.org/2005/08/addressing">
    <SOAP-ENV:Body>
        <tds:GetNetworkInterfaces></tds:GetNetworkInterfaces>
    </SOAP-ENV:Body>
</SOAP-ENV:Envelope>

DeviceService - Calltype : info, Data : Attempting to bind to /onvif/device_service
DeviceService - Calltype : info, Data : Trying Device from path 
DeviceService received request GetNetworkInterfaces
DeviceService - Calltype : replied, Data : <?xml version="1.0" encoding="utf-8"?>
<soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" 
    xmlns:i0="http://www.onvif.org/ver10/device/wsdl" 
    xmlns:tns="http://tempuri.org/" 
    xmlns:tm="http://microsoft.com/wsdl/mime/textMatching/" 
    xmlns:tds="http://www.onvif.org/ver10/device/wsdl" 
    xmlns:tt="http://www.onvif.org/ver10/schema">
    <soap:Body>
        <tds:GetNetworkInterfacesResponse xmlns:tds="http://www.onvif.org/ver10/device/wsdl" 
            xmlns="http://www.onvif.org/ver10/device/wsdl">
            <tds:NetworkInterfaces token="lo" 
                xmlns:tt="http://www.onvif.org/ver10/schema">
                <tt:Enabled>true</tt:Enabled>
                <tt:Info>
                    <tt:Name>lo</tt:Name>
                    <tt:HwAddress>00-00-00-00-00-00</tt:HwAddress>
                    <tt:MTU>1500</tt:MTU>
                </tt:Info>
                <tt:IPv4>
                    <tt:Enabled>true</tt:Enabled>
                    <tt:Config>
                        <tt:Manual>
                            <tt:Address>127.0.0.1</tt:Address>
                            <tt:PrefixLength>8</tt:PrefixLength>
                        </tt:Manual>
                        <tt:DHCP>false</tt:DHCP>
                    </tt:Config>
                </tt:IPv4>
            </tds:NetworkInterfaces>
            <tds:NetworkInterfaces token="eth0" 
                xmlns:tt="http://www.onvif.org/ver10/schema">
                <tt:Enabled>true</tt:Enabled>
                <tt:Info>
                    <tt:Name>eth0</tt:Name>
                    <tt:HwAddress>b8-27-eb-d0-b0-81</tt:HwAddress>
                    <tt:MTU>1500</tt:MTU>
                </tt:Info>
                <tt:IPv4>
                    <tt:Enabled>true</tt:Enabled>
                    <tt:Config>
                        <tt:Manual>
                            <tt:Address>192.168.123.110</tt:Address>
                            <tt:PrefixLength>24</tt:PrefixLength>
                        </tt:Manual>
                        <tt:DHCP>false</tt:DHCP>
                    </tt:Config>
                </tt:IPv4>
            </tds:NetworkInterfaces>
        </tds:GetNetworkInterfacesResponse>
    </soap:Body>
</soap:Envelope>

--

prometrix, 2019-01-04:
The problem was that the response included the 'lo' interface.
The fix was device_service.js(326)
"if (nwifs[nwif][addr].family === 'IPv4' **&& nwif !== 'lo'**) {"

RogerHardiman, 2019-01-04:
Hi.
Thanks for the bug fix. 
I was just about to email and say that I've not got an up to date Milestone install here right now so it may take a while to fix (current dev laptop has Win 10 Home and not Win 10 Pro so cannot install the Milestone server)

But I see you have got there with a fix which is great and I'm just about to add it to GitHub.

Thanks and let me know if you find any other issues as the last tests I did with Milestone were at least 14 months ago

closed 2019-01-04 by RogerHardiman

RogerHardiman, 2019-01-15:
Hi. Did the fix on github work? There is another user reporting milestone issues so wanted to check

RogerHardiman, 2019-01-16:
Sorry, when I copied over your change I was testing on a Mac.
The loopback interface was 'lo0' and not 'lo' so I fixed it on the Mac but not on the Pi.
Change now committed.