BreeeZe/rpos #54: getting errors trying to install and configure

pacman8521, 2019-01-07:
RogerHardiman if you are out there I could use some help. I've gone through the rpos install twice and am having a few minor problems. I feel as though all of the components are installed correctly but am unable to view the stream. I'm stuck. My end goal is to set up a RTSP camera on raspberry and link it into my Geovision Recording server as a RTSP feed across my network. I am unable to browse the feed in ONVIF Device Manager.

RogerHardiman, 2019-01-07:
I need the following
a) what hardware you have (and the camera type)
b) cut and paste all the things that RPOS prints on the screen
c) tell me what changes you have edited in rposConfig.json

pacman8521, 2019-01-07:
I am running Debian 9.6 - Stretch on a Raspberry Pi 3 Model B v1.2.  
The Camera is the Raspberry PI Camera Module v@ - 8 Mg, 1080p.  

**I modified rposConfig.json as follows:**
 "IpAddress" : "10.1.1.154", (same as the wireless adaptor IP addr)
  "Username" : "root",
  "Password" : "root",

**JSON is reporting the following error:**
pi@camera1-rpi:~ $ node rpos.js
module.js:339
    throw err;
    ^

Error: Cannot find module '/home/pi/rpos.js'
    at Function.Module._resolveFilename (module.js:337:15)
    at Function.Module._load (module.js:287:25)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:134:18)
    at node.js:961:3

Thank  you!
Tony

RogerHardiman, 2019-01-07:
from the RPOS directory, do
ls -l

then I can check you have compiled rpos.ts into rpos.js

pacman8521, 2019-01-07:
pi@camera1-rpi:~/rpos $ ls -l
total 676
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 bin
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 cpp
-rw-r--r--   1 pi pi   3055 Jan  7 15:32 ffserver_mac.conf
-rw-r--r--   1 pi pi   2698 Jan  7 15:32 gulpfile.js
drwxr-xr-x   2 pi pi   4096 Jan  7 15:38 lib
-rw-r--r--   1 pi pi   1128 Jan  7 15:32 LICENSE
drwxr-xr-x 535 pi pi  20480 Jan  7 15:35 node_modules
-rw-r--r--   1 pi pi   1020 Jan  7 15:32 package.json
-rw-r--r--   1 pi pi 216007 Jan  7 15:35 package-lock.json
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 python
-rw-r--r--   1 pi pi 262709 Jan  7 15:32 raspberry_pi_missile_launcher.jpg
-rw-r--r--   1 pi pi   8014 Jan  7 15:32 README.md
-rw-r--r--   1 pi pi   1204 Jan  7 16:44 rposConfig.json
-rw-r--r--   1 pi pi   1198 Jan  7 15:32 rposConfig.release.json
-rw-r--r--   1 pi pi   1889 Jan  7 15:32 rpos.d.ts
-rw-r--r--   1 pi pi   2133 Jan  7 16:51 rpos.js
-rw-r--r--   1 pi pi   5854 Jan  7 16:51 rpos.js.map
-rw-r--r--   1 pi pi   5601 Jan  7 15:32 RPOS.njsproj
-rw-r--r--   1 pi pi  63867 Jan  7 15:32 RPOS_PanTiltHAT.jpg
-rw-r--r--   1 pi pi    955 Jan  7 15:32 RPOS.sln
-rw-r--r--   1 pi pi   3338 Jan  7 15:32 rpos.ts
drwxr-xr-x   3 pi pi   4096 Jan  7 15:38 services
-rw-r--r--   1 pi pi    180 Jan  7 15:32 setup_v4l2rtspserver.sh
-rw-r--r--   1 pi pi    239 Jan  7 15:32 tsconfig.json
drwxr-xr-x   4 pi pi   4096 Jan  7 16:50 typings
-rw-r--r--   1 pi pi   1434 Jan  7 15:32 typings.json
-rw-r--r--   1 pi pi    741 Jan  7 16:53 v4l2ctl.json
drwxr-xr-x  11 pi pi   4096 Jan  7 15:49 v4l2rtspserver
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 views
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 web
drwxr-xr-x   2 pi pi   4096 Jan  7 15:32 wsdl
pi@camera1-rpi:~/rpos $

RogerHardiman, 2019-01-07:
Hi
Thanks
I can see rpos.js exists and is 2133 bytes long. that is good.

Are you sure you are in the RPOS folder when typing
 node rpos.js

The error is the same error you get when NodeJS cannot find the rpos.js file

pacman8521, 2019-01-07:
ok - I ran node rpos.js from within the rpos directory and the services started.  That's huge progress.  

Now, on my Windows pc running ONVIF Device Manager v2.2.250 I get a no endpoint listening error when browsing to 10.1.1.154.

RogerHardiman, 2019-01-07:
You need to browse to the Port - it is normally 8081
eg http://10.1.1.154:8081

pacman8521, 2019-01-07:
> When I browse to the port in a browser I get the settings page as
> expected.
>
> When I browse in the ONVIF software I get a comment type mismatch error.
> See attached.
>
> Thank you,
>
> Anthony Pacuraru
> pacman8521@gmail.com
> (215) 804-6949

Attachment not allowed....

RogerHardiman, 2019-01-07:
ok. thanks for the update.

Maybe there is a problem with ONVIF Discovery.
I did make some changes to the discovery XML recently so I may have broken something.

I'll have access to a Windows machine in 3 days time so I can try ONVIF Device Manager.
(right now I just have my Mac with me and remote login in Linux boxes)

pacman8521, 2019-01-07:
Ok thanks.

The ONVIF step is really just for validation. If I wanted to add this
camera to my production recording server what path would I use?

Also, will this setup work in headless mode?

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
(215) 804-6949

RogerHardiman, 2019-01-08:
rtsp://yourIP:8554/h264     (the RTSP port can be changed in the rposConfig.json file)

Actually there are 3 different RTSP servers. You may do better with Server 2 or Server 3 and you have to edit rposConfig to pick your RTSP server. build instructions are in the README

Or direct ONVIF is    http://yourIP:8081/onvif/device_service

pacman8521, 2019-01-08:
Thank you!
I'll confirm settings and functionality.

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
(215) 804-6949

pacman8521, 2019-01-08:
I've been tweaking settings and getting the raspberry environment just how
i want it.

When I run node rpos.js I get the following:

pi@camera1-rpi:~/rpos $ node rpos.js
Read IP address 10.1.4.175 from eth0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_PI_3
HardwareId :
SerialNumber : 00000000e965f357
FirmwareVersion : 2.0.4
Use modprobe to load the Pi Camera V4L2 driver
e.g.   sudo modprobe bcm2835-v4l2
       or the uv4l driver
pi@camera1-rpi:~/rpos $

Is this an error?

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
215-804-6949

pacman8521, 2019-01-08:
I ran  sudo modprobe bcm2835-v4l2
then re-ran node rpos.js and that seems to have worked.

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
215-804-6949

pacman8521, 2019-01-08:
Moving to production this week.  How do I get this to start automatically
at boot?

The commands I need to run are:
sudo modprobe bcm2835-v4l2
cd /home/pi/rpos
node rpos.js

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
215-804-6949

RogerHardiman, 2019-01-09:
I forget exactly what I did on my Pi but just google for running a shell script at boot on Linux.

RogerHardiman, 2019-01-09:
The sudo modprobe is needed to load the Video4 Linux drivers

RogerHardiman, 2019-01-11:
What did you use in the end to start on boot?

pacman8521, 2019-01-11:
Never got it working... need to figure it out this Monday.

Thank you,

Anthony Pacuraru
pacman8521@gmail.com
(215) 804-6949

am549, 2019-01-17:
Did you get anywhere with getting it to start on boot? I'm struggling to find something that works

ucfalmi, 2019-03-09:
Perhaps you can try crontab -e and specify @reboot?
This is what I intend to do.