BreeeZe/rpos #55: Cannot connect to camera ip on port 8081

bhizzle, 2019-01-12:
Hello, I believe I have everything installed correctly after having issues with node.  I load the camera driver per instructions and run rpos.js.  I cannot connect to the interface using 127.0.01:8081 or 192.168.1.xx on a different computer on the network.  I really can't confirm that the service is running, not sure how to.  Is there a step i'm missing here?

bhizzle, 2019-01-13:
ok, so I decided to start back from scratch.

I'm thinking this is the problem.
Any advice would be great!!

[image]

bhizzle, 2019-01-14:
Hi Roger, thanks for the reply.  I was able to run the install using that command and I nolonger received those warnings.  However, I am back at where I started here.  Everything installed, rpos not starting the service.  Here is a picture of what happens when I run the rpos.js file.  Ignore the random updates and installs I ran above... the result was the same before I started retracing steps in the readme and then searching the web for update commands.  I wonder if this has to do with hardware differences?  Im running this on a pi zero w.  I didn't think that would matter, but maybe it does.

bhizzle, 2019-01-14:
[image]

bhizzle, 2019-01-14:
ok.  I am using the 5mp OV5647 camera.  The one that looks like Johnny 5 with the IR blasters on the side.

The output of dmesg is bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.

using the last command you sent, I get Failed to open /dev/video0: No such file or directory

RogerHardiman, 2019-01-14:
does the camera work fine with raspivid

bhizzle, 2019-01-14:
I haven't tried using another application.

bhizzle, 2019-01-14:
Though I'm skeptical about this ribbon cable. Going to swap it out

RogerHardiman, 2019-01-14:
best place to start will be the standard Raspberry Pi tools for using cameras.
they are raspivid and raspistill and there are lots of resources online.

These apps talk direct to the hardware.

There is a wrapper that makes the hardware appear as a Video4Linux device and that is what the modprobe installed and what RPOS uses.

So using the raspivid and raspistill tools test at the most simple level

bhizzle, 2019-01-14:
Perfect.  That's basic principle right?  Make sure the hardware is fully functioning haha.  I appreciate your quick responses and hopefully this thread will help others with this.  1 suggestion... maybe adding a line saying camera not found would be a nice addition to the next iteration.  I wouldn't have bugged you if I knew why the service wasn't starting.  But I do greatly appreciate your prompt reply.  I will update this as I get things working (or not) lol.

RogerHardiman, 2019-01-14:
we could make some changes to RPOS to improve the error messages, would need to detect if bcm2835-v4l is installed and then if /dev/video0 is missing the error is the camera is not found

bhizzle, 2019-01-14:
It's running now. It was the ribbon cable.  So I can get to the configuration page and change settings through the IP address.  It won't pull into Milestone yet, but that is probably me.  I haven't verified an image yet, so once I do that I'll start down that rabbit hole.

RogerHardiman, 2019-01-14:
great. if you want to submit a pull request for the source code to improve error messages that would be great
Quick test    ffplay rtsp://yourip:8554/h264

bhizzle, 2019-01-14:
Ok thanks Roger.  Last question of the night.  Is it supposed to hang like this forever or am I supposed to get my prompt back at some point?

The rtsp stream does not connect.  I had tried that already and just verified its still a no go.

[image]

RogerHardiman, 2019-01-14:
You don't get the prompt pack.
I spawn it with      node rpos.js &

Have a look at the nohup command in linux too

bhizzle, 2019-01-15:
Worked on this a bit more tonight.  I have this mostly working.  The camera is now live streaming upon running the rpos.js and also through raspivid.  I can access the camera settings page on port 8081 and using VLC I can watch the live stream from :8554/h264.  However, I CANNOT get milestone to pull in the camera successfully.  It detects it, finds the correct model info, but refuses to add the camera.  Here are the screen pics.  Have any idea where to start with this?  

[image]

[image]
[image]

RogerHardiman, 2019-01-15:
It did work with milestone once and someine did a bug fix recently for their milestone system.
But rpos does not implement all  of the ONVIF spec so it could be an rpos bug.

What version of milestone are you on?

RogerHardiman, 2019-01-15:
Actually can you open a new issue with a title of Milestone plus the milestone version . Then I can track the milestone problem.  Thanks.

prometrix, 2019-01-16:
I got the exact same problem before I made the fix ( that should be in the code now ) problem is that Milestone don't accept a NIC with localhost and no real mac-address.

RogerHardiman, 2019-01-16:
Hi.
I have a milestone bug fix that I did not implement properly. I'll try and fix it later today.

RogerHardiman, 2019-01-16:
Commit has been made. Please get the latest source from GitHub, rebuild it and try it with Milestone.

@prometrix had sent in a fix to look for loopback network interfaces. I'd not realised the Pi uses 'lo' whereas the Mac uses 'lo0'.

bhizzle, 2019-01-16:
Awesome.  I didnt have a chance to work on anything last night but I will
give it a go tonight..

bhizzle, 2019-01-17:
Works PERFECT!  Thank you so much!

closed 2019-01-17 by bhizzle

RogerHardiman, 2019-01-17:
Great