BreeeZe/rpos #57: Unable to install

am549, 2019-01-16:
Hi

I'm having issues getting this installed. Running the latest release of Raspian and when I run the npm install from the rpos directory I get the following items logged. I'm running everything under su.

[image]

Any ideas?

RogerHardiman, 2019-01-16:
not sure what's wrong.
I can see Permisison Denied errors in the screenshot.

I don't built it as root.
I just build it as a normal user.

am549, 2019-01-16:
Looks like a reboot and an install as normal user has fixed it. Was getting similar issues logged in as the default user before rebooting

RogerHardiman, 2019-01-16:
Hi
Last time I did a fresh install I had a feeling the SerialPort module failed first time, but worked 2nd time.
But I then forgot all about it.

So maybe there is an issue there - but I don't fancy making a clean install to test with.

Anyway glad you got it working

closed 2019-01-17 by RogerHardiman