BreeeZe/rpos #58: PTZ emulation/status

strarsis, 2019-01-18:
It would be awesome if this ONVIF server can show a virtual camera with its current orientation, changed by PTZ commands. Debugging PTZ-based apps would be made so much easier.

RogerHardiman, 2019-01-18:
It would be awesome. I'd often thought about a 3D generated world with the viewport turned into an RTSP stream. Would be great to do Pan/Tilt/Zoom inside Minecraft (or classic Doom) 

I've never seen anything that can do it though.

If you want to write something (I can see you are a programmer as you are asking about debugging) then I'd be interested to incorporate it.
You'd need a RTSP server (we have 3 RTSP servers already so making a 4th is no problem) and need to modify PTZDriver.ts to pass PTZ events into your 3D model

strarsis, 2019-01-18:
@RogerHardiman: How to derive a new RTSP server? Is this ONVIF server based on a RTSP server JS class?

closed 2019-02-07 by RogerHardiman