BreeeZe/rpos #59: rtspServer error and no video

mxplea10, 2019-01-18:
I installed rpos on a Raspberry Pi and although it is discovered by the Dahua NVR, the video is not generated.  After approximately 30 seconds, the NVR stops trying to access the Raspberry Pi camera module.  The current setup is the following;

Raspbian 9.6 - Stretch
Raspberry Pi 3 Model B Rev 1.2
Raspberry Pi Camera Module V2 - 8 Megapixel, 1080p

*** Changes made to rposConfig.json ***

"IpAddress" : "192.168.1.147",  (this is the Raspberry Pi IP address)
"Username" : "admin",
"Password" : "admin",
"RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
"RTSPPort" : 554,  (get the same results even when set to default 8554)
"RTSPServer" : 2,
"Model" : "3 B",
"HardwareId" : "BCM2835"

pi@raspberrypi:~/rpos $ node rpos.js
Read IP address 192.168.1.147 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_PI_3
HardwareId : BCM2835
SerialNumber : 00000000d05f7f42
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.147:8081/
Binding DeviceService to http://192.168.1.147:8081/onvif/device_service
Binding MediaService to http://192.168.1.147:8081/onvif/media_service
Binding PTZService to http://192.168.1.147:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.147:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started
rtspServer error: Error: spawn v4l2rtspserver ENOENT
(node:1232) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

********
Additionally, the following error generated on a few occasions although the username and password was correct; 

Invalid username/password with GetVideoEncoderConfiguration

Should the username and password in the rposConfig.json file be the same as the Raspberry Pi username and password?  Also, the camera settings (http://192.168.1.147:8081/) have been set back to the default settings since none of the changes helped with this issue.

If anyone can tell me how to fix this issue, I would appreciate it.

RogerHardiman, 2019-01-18:
I've never used a Dahua NVR before so you may have found a bug or issue.

It looks like you may have issues opening the Pi camera
  rtspServer error: Error: spawn v4l2rtspserver ENOENT

The username and password in rposConfig are independent of the Linux level username/password.

So all I can suggest is checking that /dev/video0 is created by the bcm2835-v4l2 driver

mxplea10, 2019-01-18:
Thinks for getting back so quickly.  The /dev/video0 is there or, at least, it seems to be.

pi@raspberrypi:~/rpos $ ls /dev/video0
/dev/video0

I did notice that sudo modprobe bcm2835-v4l2 does generate an output when the command is ran.  I verified that the camera works by following this link, https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/5 

pi@raspberrypi:~/rpos $ sudo modprobe bcm2835-v4l2
pi@raspberrypi:~/rpos $ node rpos.js
Read IP address 192.168.1.147 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_PI_3
HardwareId : BCM2835
SerialNumber : 00000000d05f7f42
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.147:8081/
Binding DeviceService to http://192.168.1.147:8081/onvif/device_service
Binding MediaService to http://192.168.1.147:8081/onvif/media_service
Binding PTZService to http://192.168.1.147:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.147:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started
rtspServer error: Error: spawn v4l2rtspserver ENOENT

The error is still generated.  I'll keep trying to solve.

mxplea10, 2019-01-19:
Just an update...  I'm fairly certain that the bcm2835-v4l2 driver created /dev/video0 and that it's not the issue.

pi@raspberrypi:~/rpos $ ls -la /dev/video0
crw-rw----+ 1 root video 81, 0 Jan 18 18:41 /dev/video0

pi@raspberrypi:~/rpos $ lsmod | grep video
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  1 videobuf2_vmalloc
videobuf2_v4l2         24576  1 bcm2835_v4l2
videobuf2_core         45056  2 bcm2835_v4l2,videobuf2_v4l2
videodev              184320  4 v4l2_common,videobuf2_core,bcm2835_v4l2,videobuf2_v4l2
media                  32768  1 videodev

The issue seems to be with the rtspServer communicating with the Dahua NVR.  If there's a bug or issue with the rtspServer communicating with the Dahua NVR, what's the procedure for finding a fix?  

Also, can you or someone else recommend other checks or tests to try to solve this problem?  I'd like to test the Raspberry Pi Onvif Server configuration with another NVR.  Are there Windows and Linux NVR software packages that are known to work?  What hardware and software are others using with rpos?

RogerHardiman, 2019-01-19:
You asked about testing. You can can check the RTSP stream with  
    ffplay rtsp://192.168.1.147:8554/h264      (or I see you switched to port 554 in your config)

mxplea10, 2019-01-20:
Thanks for pointing me in the right direction.  The v4l2rtspserver application wasn't installed.  Although I had executed the command, _sh setup_v4l2rtspserver.sh_, to install v4l2rtspserver and didn't get errors, the application wasn't loaded.  I had to install it manually.  I also found that the RTSPPort had to be set at the default 8554 in order to view the video with the Dahua NVR and, also, SmartPSS for Windows. 

Referencing **How to Install on a Raspberry Pi:** STEP 6 - EDIT CONFIG, where are the settings to enable a basic ONVIF/RTSP Gateway and to hard code an IP address in the ONVIF SOAP messages?  What would be the ONVIF/RTSP Gateway device and should the ONVIF SOAP message IP address be dedicated or a specific device's IP address?

Also, where is the Resolution setting for the camera?  I didn't see it on the Web Page run with rpos.

RogerHardiman, 2019-01-20:
Hi
a) so is there a problem in setup_v4l2rtspserver and I've not executed it recently?

b) To run as a gateway you put in the full RTSP address in rposConfig.json. In theory this could be an RTSP server on a totally different machine. But ONVIF clients (especially in their NAT support) often re-write the RTSP IP address. So the RTSP source has to be the RPOS machine. You then need to run a RTSP Proxy on the same machine as the Pi, The Live555 project has an RTSP proxy.

c) Resolution is currently hard coded

mxplea10, 2019-01-20:
Hello,

**A.)**  I first tried to executed setup_v4l2rtspserver.sh from the Home directory and received errors.  Then, I executed it in the rpos directory but the command didn't generate an output.  This happened on the first and second configuration attempt.  I thought I might have missed something so I configured it again with the same results.  The server load when I used the [mpromonet/v4l2rtspserver](https://github.com/mpromonet/v4l2rtspserver/wiki/Setup-on-Pi) configuration.

**B.)**  I didn't see a setting for the RTSP gateway and am not sure if it needs to be configured in my setup.  My plan is to configure four (4) to eight (8) Raspberry Pi Camera Modules and integrate them into my current security system.  Should this be configured for my setup?

**C.)**  By hard coded, do you mean that the Resolution can't be changed at this time?  What's the default Resolution?

I configured the Raspberry Pi Onvif camera in the Dahua NVR.  However, the recording are continuous although the camera is configured to record motion detection only in the NVR.  Is there a motion detection setting in the Raspberry Pi Onvif Server configuration?  I haven't been able to find it.

I hope I'm not asking too many questions and providing too much detail.  I hope this will help someone else as well.

RogerHardiman, 2019-01-20:
a) Will try and use mpromonet's instructions

b) you only need the RTSP gateway when you are using a different RTSP server. If you are using 4 to 8 Pis then you won't need the gateway setting.

c) Resolution starts as 1280x720 I think. You can change it via ONVIF commands but it currently does not store the new resolution. It needs someone to make some code changes to save the resolution.

d) RPOS has no motion detection function

pacman8521, 2019-12-08:
Hello, MXPLEA10, did you ever get the rtsp camera server running?  Which service did you end up going with?
I feel like I'm close but I'm having trouble getting everything running.  Do I need to run the server as a service?  When I try to run the v4l2rtspserver I get a command not found error...really want to deploy 2-4 raspberry pi cameras but it needs to be sustainable...

pacman8521, 2019-12-08:
For now I'm using rtsp server 1 and I'm up and running.  Concerned about the statement that this server can be unstable.  Also, how can I clone my entire Raspberry sd card to a new one so that I don't have to start from scratch for each camera?

pacman8521, 2019-12-08:
Here is the error I get when using server 2 or server 3: 
pi@camera1-rpi:~/rpos $ node rpos.js
Read IP address 192.168.160.215 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B+_PI_3
HardwareId : 
SerialNumber : 00000000b87b207b
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.160.215:8081/
Binding DeviceService to http://192.168.160.215:8081/onvif/device_service
Binding MediaService to http://192.168.160.215:8081/onvif/media_service
Binding PTZService to http://192.168.160.215:8081/onvif/ptz_service
Binding ImagingService to http://192.168.160.215:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started
rtspServer error: Error: spawn v4l2rtspserver ENOENT

mxplea10, 2019-12-08:
Hello, pacman8521, I got the service running without errors.  I test 3 cameras for several months but decided against using them. I could view the cameras with my surveillance system but couldn't save the recordings.  I used MotionEye to saving the recordings to a network drive.  I don't specifically remember what cause all the errors but I did have a note that errors were caused because the Raspberry Pi Camera Module wasn't configured.  To clone your SD card, check out this tutorial, [How to Backup your Raspberry Pi SD Card](https://pimylifeup.com/backup-raspberry-pi/).

Here's my basic configuration from my notes. 

sudo modprobe -v bcm2835-v4l2

************ 

**** File: v4l2rtspserver.service
[Unit]
Description=V4L2 RTSP server
After=network.target
StartLimitIntervalSec=0

[Service]
Type=simple
Restart=always
RestartSec=1
User=pi
ExecStart=/home/pi/startrtspserver.sh

[Install]
WantedBy=multi-user.target

************ 

**** File: /etc/systemd/system/rtsp.service
[Unit]
Description=rtsp service
After=v4l2rtspserver.service

[Service]
ExecStart=/home/myDIRECTORY/rtsp.sh
WorkingDirectory=/home/myDIRECTORY
Restart=always
User=myDIRECTORY

[Install]
WantedBy=v4l2rtspserver.service

sravich1, 2023-12-18:
Hi,
I am a trying to run the rpos code in Rasppi4 ..when I run the rpos.js I am getting following error
rtspServer error: Error: spawn ./bin/rtspServer ENOENT
at Process.ChildProcess._handle.onexit (internal/child_process.js:268:19)
at onErrorNT (internal/child_process.js:470:16)
at processTicksAndRejections (internal/process/task_queues.js:84:21) {
errno: 'ENOENT',
code: 'ENOENT',
syscall: 'spawn ./bin/rtspServer',
path: './bin/rtspServer',
spawnargs: [Array]
}

looks like rtsp server is unable to start...need to know what changes need to be made in addition to all the installation done.
Kindly let me know .
Thanks
Ravi