BreeeZe/rpos #61: onvif not working through openvpn address

cfdcfc, 2019-01-25:
Hi,
I am using GST-RTSP-SERVER (server option 3) on the Raspberry Pi, the network adapter is set to eth0.
on the Mac I use SecuritySpy and the address is set to same address as eth0 and the connection is built successfully. 
I tried
`sudo tcpdump -i any port 8554
and got the traffics.

However, when I change the network adapter from eth0 to tun1 (openVPN) on Pi, and on the Mac I set the address to the same address as tun1, the connection failed.
I tried
`sudo tcpdump -i any port 8554
but got nothing.

Then, I try to connect the rtsp server by tun1 address, i.e. rtsp://10.8.0.156:8554/h264 and it works. (through VLC player) 

I am wondering which part goes wrong?

rposConfig.json:
>    {
>   "NetworkAdapters" : ["tun1"],
>   "IpAddress" : "",
>   "ServicePort" : 8081,
>   "Username" : "pi",
>   "Password" : "pi",
>   "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
>   "RTSPPort" : 8554,
>   "RTSPName" : "h264",
>   "MulticastEnabled"  : false,
>   "RTSPMulticastName" : "h264m",
>   "MulticastAddress"  : "224.0.0.1",
>   "MulticastPort"     : "10001",
>   "RTSPServer" : 3, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
>   "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
>   "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
>   "PTZSerialPort" : "/dev/ttyUSB0",
>   "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
>   "PTZOutputURL": "127.0.0.1:9999",
>   "PTZCameraAddress": 1,
>   "DeviceInformation" : {
>     "Manufacturer" : "Raspberry Pi",
>     "Model" : "2 B",
>     "HardwareId" : ""
>   },
>   "logLevel" : 4, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
>   "logSoapCalls" : false
> }

RogerHardiman, 2019-01-25:
In rposConfig..json, try putting in the Pi's OpenVPN IP address in
    IpAddress" : "",

eg
IpAddress" : "/10.8.0.156",

Some of the ONVIF commands the Pi sends out have to include the Pi's IP Address.
ONVIF applications like SecuritySpy then use the IP address from the ONVIF Command.
The Pi tries to work out what its IP address is but maybe it gets this wrong in your case.
Setting "IpAddress" in rposConfig.json tells RPOS what IP address to send out in ONVIF replies

cfdcfc, 2019-01-25:
I tried change the IpAddress but still no luck. When I am connecting, the PROS log nothing so I am not sure where I should start with.

Is there any other way to hardcode the Pi's IP address?

cfdcfc, 2019-01-28:
I try to connect with ONVIF Device Manager on Windows and succeed, very likely it's a bug of SecuritySpy.

RogerHardiman, 2019-01-28:
It would be worth reporting this to SecuritySpy.
I've sent them problem reports in the past and found them to be really helpful.

Roger

RogerHardiman, 2019-02-07:
Did your first post say you are setting the Pi to the same up address as the mac?