BreeeZe/rpos #62: Milestone issue

tallerken, 2019-01-29:
Hi! New to both raspian and github here. Having some issues with getting my camera into Milestone. Have everything installed (?) according to readme, but can't get it to work. Shows up and is discovered with hardware and everything in Milestone, can also connect but shows "no video". Had a couple of blips with video when I look at recording, but unfortuneately cannot identify what exacly I did differently then. If anyone can help, highly appreciated.

This is what i get when node rpos.js (launching driver with modprobe first)

> pi@raspberrypi:~/rpos $ Read IP address 10.0.212.200 from eth0
> Manufacturer : RPOS Raspberry Pi
> Model : Model_PiZeroW_Revision_1.1
> HardwareId : 
> SerialNumber : 0000000005115329
> FirmwareVersion : 2.0.4
> Starting camera settings webserver on http://10.0.212.200:8081/
> Binding DeviceService to http://10.0.212.200:8081/onvif/device_service
> Binding MediaService to http://10.0.212.200:8081/onvif/media_service
> Binding PTZService to http://10.0.212.200:8081/onvif/ptz_service
> Binding ImagingService to http://10.0.212.200:8081/onvif/imaging_service
> discovery_service started
> ptz_service started
> imaging_service started
> device_service started
> Starting Live555 rtsp server
> media_service started
 --LAST LINE, nothing more happens--

My rposConfig.json:

>   "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"], 
  "IpAddress" : "10.0.212.200",
  "ServicePort" : 8081,
  "Username" : "pi",
  "Password" : "xxxx",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 2, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "2 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}

> pi@raspberrypi:~/rpos $ npm -v
6.7.0
pi@raspberrypi:~/rpos $ node -v
v10.15.0

RogerHardiman, 2019-02-07:
Thanks for letting me know it works with RTSP server 1.
What version of Milestone are you running and which release?

RogerHardiman, 2019-03-02:
Not had any more feedback on which version of Milestone is being used.

tallerken, 2019-03-06:
Sorry for late response; 2017 R3

bgbigbrother, 2020-10-21:
Tested with 2020 R1 and it work just fine