BreeeZe/rpos #63: resolution or framerate is not controlled via ONVIF device manager

PaulHwang19, 2019-02-10:
Now, I'm trying to use rpos in my RPI board and PI cam. During my testing, I used rpos control webpage and ONVIF device manager to control camera. The streaming is OK in VLC and ONVIF device manager. 
I can control camera setting using default control webpage but I cannot find resolution or framerate in there. So I tried them in the ONVIF device manager but I failed. They are not controlled via ONVIF device manager. The resolution list is shown in the dropdown list but my selection is not applied. 
How can I solve this?

RogerHardiman, 2019-02-10:
Hi
The framerate and resolution are areas I've never worked on in RPOS. The code is still the original code from @BreeeZe 
So there may be bugs or the implementation may not be complete.

If you want make some code changes that would be great and I can review the code and merge it.
You would need to add resolution settings to the Web Page and then store the settings in the JSON v4l2 config file (where all the other settings are stored), then use those settings when the app loads.

You could then also modify the ONVIF SOAP messages to use the new resolution code.

Roger

mjn9, 2020-11-06:
how to change resolution of camera using libonvif libratu