BreeeZe/rpos #64: Stream times out & start on boot?

developer-steve, 2019-03-02:
Help would be greatly appreciated, relatively new to ONVIF streaming.

**Setup has gone well, executed commands:**
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install npm
sudo npm install -g npm@latest
sudo apt-get install git
git clone https://github.com/BreeeZe/rpos.git
cd rpos
npm install
npx gulp or ./node_modules/gulp/bin/gulp.js
sudo apt-get install liblivemedia-dev
sh setup_v4l2rtspserver.sh
sudo raspi-config (enabled camera & ssh from menu)
sudo nano /etc/modules --> add this line for load drive at boot -->  modprobe bcm2835-v4l2
node rpos.js

**node.js starts like:**
`Read IP address 192.168.1.236 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_PiZeroW_Revision_1.1
HardwareId : 
SerialNumber : 00000000xxxxxxx
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.236:8081/
Binding DeviceService to http://192.168.1.236:8081/onvif/device_service
Binding MediaService to http://192.168.1.236:8081/onvif/media_service
Binding PTZService to http://192.168.1.236:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.236:8081/onvif/imaging_service
discovery_service started
device_service started
Starting Live555 rtsp server
media_service started
ptz_service started
imaging_service started
rtspServer: Streaming on URL "rtsp://192.168.1.236:8554/h264"

(RTSP-over-HTTP tunneling is not available.)
`
**VLC:**
I open stream "rtsp://192.168.1.236:8554/h264", loads okay but stops after about 10 seconds.
The RPi Camera led lights up and goes off coinciding with the above.

**Hikvision NVR:**
The NVR discovers the camera all on its own and its a one click add.
Initially all seems fine but somehow the stream stops and can't be recovered and just says 'no link'.
The RPi Camera led stays on all the time.
Maybe it is something with the protocol, RSP, TDP, UDP, not sure what should be selected?

**Boot:**
Also how do I get this to start on boot for headless operation? I tried a number of tutorials but none of them worked out?

Is it possible to pull the stream twice from the camera, don't need to but just noticed I needed to make sure either VLC or the NVR was off as this seemed to cause the streaming to stop also.

Any other advice appreciated.

Thanks

RogerHardiman, 2019-03-03:
Hi
ONVIF is a protocol to help CCTV recorders and Viewing software get the low level RTSP address.
In this case you have tried VLC with the RTSP address which is good.
We can ignore the ONVIF protocol bits and just focus on RTSP streaming.

RPOS supports 3 different RTSP servers. Clearly something is going wrong with the RTSP server you are using as the camera is going off after 10 seconds.

I normally change rposConfig.json to use RTSP Server Type 2

developer-steve, 2019-03-03:
Many thanks for the above help, I changed to type 2 and VLC stream seems solid now. On the downside NVR doesn't show any picture now but does connect to the camera.

Terminal is giving me this error when VLC connects but doesn't seem to be any issue:
`rtspServer: RTCPInstance::RTCPInstance error: totSessionBW parameter should not be zero!`

The picture was cropped so I updated resolution to 1640x1232 and fps to 5 and I'm happy with this picture.

Is it possible that some of the codec controls are causing issues with the NVR, can we make it as basic as possible? I changed h264 profile to  baseline and also enabled repeat_sequence_header but they did not seem to help, any pointers here?

Also start on boot seems almost impossible, I tried rc.local, crontab -e, sudo crontab -e files, loading the .js through a .sh in all the above but just no luck, anybody have this working on a RPI?

Thanks

RogerHardiman, 2019-03-04:
Glad to hear Type 2 for the RTSP server works. Maybe I will drop the original RPOS RTSP server and just including a build of the one from mpromonet.

I've seen that 'should not be zero' message before but never looked into it. It applies to RTCP (the data channel that has stats about the video stream and not the video stream itself)

As for the NVR, I've never tried it with a Hik recorder before so don't know what is up.
There is a Hik NVR at work so I'll have to try it some time.
It could be that Hik calls one of the ONVIF commands I never implemented.

As for startup I think I used something like this in /etc/rc.local
  sudo modprobe bcm2835-v4l2
  nohup node /home/roger/rpos/rpos.js &

WP-tim, 2019-06-08:
FWIW  my setup is Raspberry Pi Zero with PI Camera, streaming to Amazon Fire TV Cube.
ONVIF detection doesn't work, but using the RTSP port directly does. (good enough for me)

However, FYI: I began with the on-board (aka native) WIFI of the PI Zero. This results in drop out every minute - the result being mostly blank screen, and a few seconds of video.

When I used a USB RJ45 dongle, the drop out are only 4 or 5 times an hour ( though still around a minute duration).

The CPU is not over utilized.  I'll try a Pi 3B+ once that host is available (currently tied up on another project)
.......................................................................................................
top - 17:10:55 up 20:32,  2 users,  load average: 0.36, 0.29, 0.27
Tasks:  84 total,   1 running,  51 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.2 us,  6.2 sy,  0.0 ni, 82.9 id,  0.0 wa,  0.0 hi,  4.7 si,  0.0 st
KiB Mem :   378172 total,   160644 free,    70108 used,   147420 buff/cache
KiB Swap:   102396 total,   102396 free,        0 used.   255588 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
 1743 laudix    20   0   26144  13324  12552 S **11.9**  3.5   7:16.48 v4l2rtspserver
 1685 laudix    20   0    8104   3304   2792 R  2.9  0.9   1:30.09 top
    7 root      20   0       0      0      0 S  0.3  0.0   0:38.31 ksoftirqd/0
   39 root       1 -19       0      0      0 S  0.3  0.0   1:49.52 vchiq-slot/0
  247 root      20   0   27592   1368   1228 S  0.3  0.4   0:51.63 rngd
 1664 laudix    20   0   11648   4376   3464 S  0.3  1.2   0:06.84 sshd
 1781 root      20   0       0      0      0 I  0.3  0.0   0:04.20 kworker/u2:1-m

Littifus, 2021-04-17:
Hi,

I know this is an old thread but I'm looking for a biut of help to complete the build.  I've followed the instructions and retried several times, I get as far as media service started and it just hangs.  Is this a common issue or is there something I've missed?  I also, first time round got some contention on port usage. Here is the output:

pi@raspberrypi:~/rpos $ node rpos.js
Read IP address 192.168.1.138 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_PiZeroW_Revision_1.1
HardwareId :
SerialNumber : 00000000ac15d911
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.138:8081/
Binding DeviceService to http://192.168.1.138:8081/onvif/device_service
Binding MediaService to http://192.168.1.138:8081/onvif/media_service
Binding PTZService to http://192.168.1.138:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.138:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started

Looks like the media service isn't starting fully.  I can get to the Web Page that runs with rpos http://<CameraIP>:8081 and seems to work but no video feed.  The hw works using other solutiuons.  Thanks

EduardoRonchi, 2021-04-25:
Any updates?? @Littifus my situation is the same as yours. I opened an issue, but no answers so far: https://github.com/BreeeZe/rpos/issues/106