BreeeZe/rpos #66: RTSP Server 3 Option - gstreamer issues

ucfalmi, 2019-03-09:
Hi - I've worked through the install using a Raspbi Zero W with the NoIR camera module.  My objective is to move away from flaky VLC streaming to something more elegant - I'm going to pick up the stream with SecuritySpy.

I'm not very experienced with this sort of thing, so excuse me if this is a dumb question.  How can I fix this error in the python that refers to the get-rtsp-launch?  I'm pretty confident that the install went OK, but this has me stumped.

The readme suggests that it's an option on the "RTSP Server" line but there are only comments for options 1 and 2, so I kind of assumed that it was 3.

```
pi@raspicam:~/rpos $ node rpos.js
Read IP address 192.168.0.134 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_PiZeroW_Revision_1.1
HardwareId : 
SerialNumber : 00000000fbfaae5f
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.0.134:8081/
Binding DeviceService to http://192.168.0.134:8081/onvif/device_service
Binding MediaService to http://192.168.0.134:8081/onvif/media_service
Binding PTZService to http://192.168.0.134:8081/onvif/ptz_service
Binding ImagingService to http://192.168.0.134:8081/onvif/imaging_service
discovery_service started
device_service started
Starting Live555 rtsp server
media_service started
ptz_service started
imaging_service started
rtspServer: Traceback (most recent call last):

rtspServer:   File "./python/gst-rtsp-launch.py", line 44, in <module>
    import gi
ImportError: No module named gi

rtspServer exited with code: 1
```

EDIT:  Just running through the install of this now:  https://www.npmjs.com/package/gi

RogerHardiman, 2019-04-07:
The gstreamer work is RTSP Server option 3.
It was contributed by another developer and I have had it working. I recall there were lots of dependencies.

johnnyxwan, 2020-01-10:
Not sure if you still need help. The README lacks this instruction. I hope this help.
`sudo apt-get install python-gi gir1.2-gst-plugins-base-1.0 gir1.2-gst-rtsp-server-1.0`

ucfalmi, 2020-01-10:
Awesome, thanks @johnnyxwan - much appreciated.