BreeeZe/rpos #67: Unauthorized access vulnerability

xfgryujk, 2019-04-02:
[This callback](https://github.com/BreeeZe/rpos/blob/2d8fdee188fc89d2c5644f8b189f715299733db5/lib/SoapService.ts#L73) won't be called if the request doesn't include any header

[image]

[image]

RogerHardiman, 2019-04-07:
thanks for the issue report.
I'm not able to do any RPOS work for few weeks. Do you have some patches I can apply?

xfgryujk, 2019-04-07:
> thanks for the issue report.
> I'm not able to do any RPOS work for few weeks. Do you have some patches I can apply?

According to [this](https://github.com/BreeeZe/node-soap/blob/961a8933214835e99c052393ffda1b3f07fda0d7/lib/server.js#L210), maybe you can change 'headers' to 'request'. But I have not tested yet