BreeeZe/rpos #68: ptz rest api

utya1988, 2019-05-02:
Hello. I have cam xiaami dafang with hack (https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks). This cam doesn't work with onvif. But have rtsp stream and ptz control by restapi command. Can i use your project with rtsp stream from this cam and have ptz control using some proxy ptz->restAPI? Can i configure rpos for using ptz/restapi mode?
Thanks

RogerHardiman, 2019-05-02:
Yes that would work - although there may be one problem.

So in RPOS you would use rposConfig.json to point at the RTSP server inside the camera.
RPOS would then handle the ONVIF Discovery and the ONVIF PTZ.
In the RPOS source is a PTZ driver which converts ONVIF PTZ into several different formats (a Raspberry Pi Pan/Tilt Motor system, into Pelco protocol, into Sony VISCA protocol). You would need to edit this file to generate PTZ commands for your camera.

So that should work.

The only problem could be if the CCTV viewing software is unhappy sending ONVIF commands to one IP address and getting RTSP streams from another.
Some CCTV software changes the IP address as part of the software to work around NAT
Eg a camera is 192.168.0.2
You access it via the internet and do ONVIF commands to Public IP address 66.55.44.33
When the CCTV software gets the RTSP address it may say "rtsp://192.168.0.2" (ie the local address) and CCTV software changes it to rtsp://66.55.44.33 
So your software may modify the RTSP address and not work

The fix for that is to look at the Live555 project. They have a RTSP proxy.
You can run that on the Raspberry Pi.
Then the ONVIF and the RTSP stream are both generated from the Pi.

If you get something working please let me know. Would be great to include any improvements into RPOS

utya1988, 2019-05-02:
thanks for reply.

> Some CCTV software changes the IP address as part of the software to work around NAT
> Eg a camera is 192.168.0.2
> You access it via the internet and do ONVIF commands to Public IP address 66.55.44.33 [...]
i will use this system in local network, so i think it's not problem for me. but thanks for advice.

> So in RPOS you would use rposConfig.json to point at the RTSP server inside the camera.
> RPOS would then handle the ONVIF Discovery and the ONVIF PTZ.
> In the RPOS source is a PTZ driver which [...]

Could you explain me, i must edit this config format?

https://github.com/BreeeZe/rpos/blob/master/services/ptz_service.ts
https://github.com/BreeeZe/rpos/blob/master/services/stubs/ptz_service.js
https://github.com/BreeeZe/rpos/blob/master/lib/PTZDriver.ts

I want deploy project on raspberry and get stream from ip cam.

RogerHardiman, 2020-05-11:
You need to edit the file PTZDriver.ts to use your REST API
(there is no need to change ptz_service.ts or ptz_service.js)

closed 2020-05-11 by RogerHardiman