BreeeZe/rpos #70: missing file

WP-tim, 2019-06-05:
Just installed rpos, and have an error on step #8
"node rpos.js"

that file "rpos.js" does not exist. 
It is not in the ~/rpos directories, nor do I see it here in the repository.

Did my install fail somewhere?

WP-tim, 2019-06-05:
arbitrarily trying a similarly named file also fails

`laudix@RPi0-CameraUpstairs:~/rpos $ node rpos.ts
/home/laudix/rpos/rpos.ts:29
import http = require("http");
^^^^^^

SyntaxError: Unexpected token import
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:616:28)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3
`

WP-tim, 2019-06-05:
in #54 @RogerHardiman says "check you have compiled rpos.ts into rpos.js"

here is my listing

`laudix@RPi0-CameraUpstairs:~/rpos $ ls -l
total 680
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 bin
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 cpp
-rw-r--r--   1 laudix laudix   3055 Jun  3 22:17 ffserver_mac.conf
-rw-r--r--   1 laudix laudix   2698 Jun  3 22:17 gulpfile.js
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 lib
-rw-r--r--   1 laudix laudix   1128 Jun  3 22:17 LICENSE
drwxr-xr-x 587 laudix laudix  20480 Jun  4 18:09 node_modules
-rw-r--r--   1 laudix laudix   1019 Jun  4 18:09 package.json
-rw-r--r--   1 laudix laudix 241659 Jun  4 18:10 package-lock.json
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 python
-rw-r--r--   1 laudix laudix 262709 Jun  3 22:17 raspberry_pi_missile_launcher.jpg
-rw-r--r--   1 laudix laudix   8088 Jun  3 22:17 README.md
-rw-r--r--   1 laudix laudix   1206 Jun  4 21:02 rposConfig.json
-rw-r--r--   1 laudix laudix   1198 Jun  3 22:17 rposConfig.release.json
-rw-r--r--   1 laudix laudix   1889 Jun  3 22:17 rpos.d.ts
-rw-r--r--   1 laudix laudix   5601 Jun  3 22:17 RPOS.njsproj
-rw-r--r--   1 laudix laudix  63867 Jun  3 22:17 RPOS_PanTiltHAT.jpg
-rw-r--r--   1 laudix laudix    955 Jun  3 22:17 RPOS.sln
-rw-r--r--   1 laudix laudix   3338 Jun  3 22:17 rpos.ts
drwxr-xr-x   3 laudix laudix   4096 Jun  3 22:17 services
-rw-r--r--   1 laudix laudix    180 Jun  3 22:17 setup_v4l2rtspserver.sh
-rw-r--r--   1 laudix laudix    239 Jun  3 22:17 tsconfig.json
-rw-r--r--   1 laudix laudix   1434 Jun  3 22:17 typings.json
drwxr-xr-x  12 laudix laudix   4096 Jun  4 18:25 v4l2rtspserver
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 views
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 web
drwxr-xr-x   2 laudix laudix   4096 Jun  3 22:17 wsdl
`

WP-tim, 2019-06-05:
I tried rerunning the install; still no file afterwards

laudix@RPi0-CameraUpstairs:~/rpos $ npm install
npm WARN gulp-typescript@5.0.1 requires a peer of typescript@~2.7.1 || >=2.8.0-dev || >=2.9.0-dev || ~3.0.0 || >=3.0.0-dev || >=3.1.0-dev || >= 3.2.0-dev || >= 3.3.0-dev but none is installed. You must install peer dependencies yourself.

up to date in 147.386s

RogerHardiman, 2019-06-05:
The .ts file is written on Typescript. The install instructions have steps to compile (transpile) the Typescript into Javascript to give you the node.js file. 

That is covered in Step 4 in the Readme file.

WP-tim, 2019-06-06:
so. I wiped the SD card and reloaded the image (using a RPi Zero)
`uname -a`
`Linux raspberrypi 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux`

I ran through the instruction, and after step 4, still no rpos.js.

thoughts? suggestions?

laudix@RPi0-CameraUpstairs:~/securitycamera/rpos $ ls -l
total 652
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 bin
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 cpp
-rw-r--r--   1 laudix laudix   3055 Jun  5 19:58 ffserver_mac.conf
-rw-r--r--   1 laudix laudix   2698 Jun  5 19:58 gulpfile.js
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 lib
-rw-r--r--   1 laudix laudix   1128 Jun  5 19:58 LICENSE
drwxr-xr-x 535 laudix laudix  20480 Jun  5 20:08 node_modules
-rw-r--r--   1 laudix laudix   1020 Jun  5 19:58 package.json
-rw-r--r--   1 laudix laudix 216511 Jun  5 20:09 package-lock.json
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 python
-rw-r--r--   1 laudix laudix 262709 Jun  5 19:58 raspberry_pi_missile_launcher.jpg
-rw-r--r--   1 laudix laudix   8088 Jun  5 19:58 README.md
-rw-r--r--   1 laudix laudix   1198 Jun  5 19:58 rposConfig.json
-rw-r--r--   1 laudix laudix   1198 Jun  5 19:58 rposConfig.release.json
-rw-r--r--   1 laudix laudix   1889 Jun  5 19:58 rpos.d.ts
-rw-r--r--   1 laudix laudix   5601 Jun  5 19:58 RPOS.njsproj
-rw-r--r--   1 laudix laudix  63867 Jun  5 19:58 RPOS_PanTiltHAT.jpg
-rw-r--r--   1 laudix laudix    955 Jun  5 19:58 RPOS.sln
-rw-r--r--   1 laudix laudix   3338 Jun  5 19:58 rpos.ts
drwxr-xr-x   3 laudix laudix   4096 Jun  5 19:58 services
-rw-r--r--   1 laudix laudix    180 Jun  5 19:58 setup_v4l2rtspserver.sh
-rw-r--r--   1 laudix laudix    239 Jun  5 19:58 tsconfig.json
-rw-r--r--   1 laudix laudix   1434 Jun  5 19:58 typings.json
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 views
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 web
drwxr-xr-x   2 laudix laudix   4096 Jun  5 19:58 wsdl

@RogerHardiman thanks for confirming where the repos.js is created

WP-tim, 2019-06-06:
Here's the output from step 4:
---------------------------
Not really sure if the warnings are expected.
....................................................

laudix@RPi0-CameraUpstairs:~/securitycamera/rpos $ npm install
npm WARN deprecated typings@0.8.1: Typings is deprecated in favor of NPM @types -- see README for more information
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
npm WARN deprecated gulp-util@3.0.8: gulp-util is deprecated - replace it, following the guidelines at https://medium.com/gulpjs/gulp-util-ca3b1f9f9ac5
npm WARN deprecated popsicle-proxy-agent@1.0.0: Use `agent` option with `popsicle` directly
npm WARN deprecated graceful-fs@3.0.11: please upgrade to graceful-fs 4 for compatibility with current and future versions of Node.js
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated natives@1.1.6: This module relies on Node.js's internals and will break at some point. Do not use it, and update to graceful-fs@4.x.
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated graceful-fs@1.2.3: please upgrade to graceful-fs 4 for compatibility with current and future versions of Node.js

> serialport@6.2.2 install /home/laudix/securitycamera/rpos/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=8.11.1 runtime=node arch=arm platform=linux)
make: Entering directory '/home/laudix/securitycamera/rpos/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1740:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:150:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated  -Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1740:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterClose(uv_work_t*)’:
../src/serialport.cpp:188:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated  -Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1740:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterFlush(uv_work_t*)’:
../src/serialport.cpp:231:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated  -Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1740:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterSet(uv_work_t*)’:
../src/serialport.cpp:285:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated  -Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1740:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:336:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated  -Wdeprecated-declarations]
   data->callback.Call(2, argv);
[... 93 lines cut ...]
  dev: 45826,
  mode: 33188,
  nlink: 1,
  uid: 1001,
  gid: 1001,
  rdev: 0,
  blksize: 4096,
  ino: 132692,
  size: 85,
  blocks: 8,
  atimeMs: 1559790514000,
  mtimeMs: 1559790514000,
  ctimeMs: 1559790514499.2815,
  birthtimeMs: 1559790514499.2815,
  atime: 2019-06-06T03:08:34.000Z,
  mtime: 2019-06-06T03:08:34.000Z,
  ctime: 2019-06-06T03:08:34.499Z,
  birthtime: 2019-06-06T03:08:34.499Z }
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/laudix/securitycamera/rpos/node_modules/serialport/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
added 717 packages from 457 contributors and audited 2792 packages in 576.433s
found 13 vulnerabilities (4 low, 2 moderate, 7 high)
  run `npm audit fix` to fix them, or `npm audit` for details

WP-tim, 2019-06-06:
Oops!

Somehow when I pasted the instructions into my Word doc I missed step 4.

So my step 4 wasn't...

Anyhow, reworking that now.

WP-tim, 2019-06-06:
Just to make this 'issue' complete...
Once I used the instructions (including step 4) I am now able to detect and view my RPI from "ONVIF Device Manager v2.2.250"

WP-tim, 2019-06-06:
Suggestion.  Modify Step 4 layout to match the other steps so the required command is not overlooked.
Specifically have "npx gulp" on a new line and indented, and prefix the comment with 'Note:"

[image]

RogerHardiman, 2019-06-07:
Glad you got it working.
Can you send over a Pull Request please and I can merge it in.

Thanks

bendrexl, 2019-11-05:
Great suggestion.
I think my pull request #76  addresses this, but feel free to review and suggest improvements there.