BreeeZe/rpos #71: Dual/multi profiles

WP-tim, 2019-06-06:
My GeoVision camera has two outputs:
- a high-res stream (I capture with Synology)
- a low-res stream (very useful remotely - especially on iPhone)

I think these streams are handled with ONVIF profiles (but I am not certain)

Has anybody experimented with multiple, simultaneous, resolutions?

RogerHardiman, 2019-06-07:
Rpos does not support multiple resolutions. rpos uses a live555 based RTSP server that uses harfware h264 encoding cia the v4l2 api. The api only supports one resolition at a time.

More recently some support for gstreamer has been added and this could be changed to support 2 streams.I don't have time to work on this but would merge in pull requests.

Roger

WP-tim, 2019-06-07:
Hmm. I chose the "mpromonet RTSP server"   because of the comment, in the instructions, "Pi users will probably prefer ..." it.

Is there a means to switch to gstreamer ?  Just install both?

(FYI just curious. I am going to re-flash my SD Card anyhow and I'll try gstreamer this weekend)

WP-tim, 2019-06-08:
I've not examined the code (I've never used nodeJS before; it's been a python journey on Raspberry)
so this may be silly, but what about recieving the full stream from the API, and having code (via CV2?) resize the frames and send them as the 2nd stream?

This might also be a pre-cursor for motion detection.

Note: I am willing to take a stab at this...  but much later this year/early next.
My current need is to feed video information to MQTT (IoT) broker, so that is priority.

shivlal12345, 2021-03-09:
> Rpos does not support multiple resolutions. rpos uses a live555 based RTSP server that uses harfware h264 encoding cia the v4l2 api. The api only supports one resolition at a time.
>
> More recently [...]

does the latest code support two RTSP streams? If not what changes are recommended?