BreeeZe/rpos #72: setup with Mozilla webthings gateway

frenchfaso, 2019-07-30:
Hy there,
I'm trying to setup a Pi Zero with rpos as an ONVIF camera managed by the Mozilla webthings gateway (which is running on a Pi 3)
I've followed the setup instructions on rpos without problems (for the Pi Zero with the official camera), and added the official Mozilla ONVIF add-on to the WebThings gateway on the Pi 3.
When configuring the ONVIF add-on the "address" and "urn" fields get automatically populated as follows:
address: http://192.168.43.242:8081/onvif/device_service
urn: urn:uuid:926810d4-68da-5dd2-bbfe-1c1f1047e9b6

which seems to indicate the gateway has found the ONVIF camera on the LAN, right?
I just have to add username and password, as configured in the rposConfig.json.

But when I go to the main page of the webthings gateway, where you can add "webthings", and try to add the ONVIF PiZero, nothing is discovered.
Am I missing something, or is rpos not compatible with the Mozilla ONVIF add-on for the webthings gateway?
Thanks in advance for your reply!

RogerHardiman, 2019-07-30:
I've not seen the Mozilla ONVIF add-on before so don't know anything about it.
I'll have to take a look at it and see what it does.

frenchfaso, 2019-07-30:
Thanks for the quick reply!
I've just read that the add-on works for sure only with 3 cameras and SHOULD work with any ONVIF Profile S compatible camera (https://github.com/mozilla-iot/onvif-adapter#readme)
Could this be the problem?

magicbycalvin, 2020-06-18:
I'd like to bump this issue. I am running into a very similar problem but am running the camera on the same Raspberry Pi 3 that is running the WebThings Gateway. I did download an ONVIF app on my phone and managed to get the video stream working, so it is very likely on Mozilla's end and not a fault of this library. I am going to also chat with Mozilla to figure out if we can determine the issue.