BreeeZe/rpos #78: video instruction for rpi3 please ???

tung256, 2019-12-31:
i tried with raspberry pi 3 B with wifi/bluetooth running Stretch and it refuses to work.  executed the last code: node rpos.js   yields nothing. could someone please make a video? thanks!

RogerHardiman, 2019-12-31:
Can you run node rpos.js and cit and paste the command line and any text or error message that is displayed please?

tung256, 2020-01-01:
here are all the commands i used during install for rpi 3 running Stretch.

sudo raspi-config
enable camera and reboot

sudo apt-get install npm

curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
sudo apt-get install nodejs

sudo npm install -g npm@latest

sudo apt install git

git clone https://github.com/BreeeZe/rpos.git
cd rpos
npm install

npx gulp

sudo nano rposConfig.json
	set ip address to ip address of pi3
	set user id
	set pw

	rtsp port 8554

	service port 8081

sudo modprobe bcm2835-v4l2
node rpos.js

rposConfig.json file
=======================
{
  "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
  "IpAddress" : "192.168.1.62",
  "ServicePort" : 8081,
  "Username" : "admin",
  "Password" : "password",
  "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
  "RTSPPort" : 8554,
  "RTSPName" : "h264",
  "MulticastEnabled"  : false,
  "RTSPMulticastName" : "h264m",
  "MulticastAddress"  : "224.0.0.1",
  "MulticastPort"     : "10001",
  "RTSPServer" : 1, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto 

selected if M$
  "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
  "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
  "PTZSerialPort" : "/dev/ttyUSB0",
  "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
  "PTZOutputURL": "127.0.0.1:9999",
  "PTZCameraAddress": 1,
  "DeviceInformation" : {
    "Manufacturer" : "Raspberry Pi",
    "Model" : "2 B",
    "HardwareId" : ""
  },
  "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : false
}

how do i get the error log you wanted? i am totally new with rpi

EduardoRonchi, 2021-04-25:
Hello, I know this is an old issue, but I am facing almost the same problem: https://github.com/BreeeZe/rpos/issues/106
any news or updtes that made it work?