BreeeZe/rpos #80: Focus Near does not work

bagrintech, 2020-01-13:
I checked the operation of rpos with the milesight VMS lite.  The command Focus Far is worked normally. But Focus Near does not work. 
It issues comes because in the PTZDriver.js  has : 

`  else if (command === 'focus') {
                if (_this.pelcod) {
                    if (data.value < 0)
                        _this.pelcod.focusNear(true);
                    else if (data.value > 0)
                        _this.pelcod.focusFar(true);
                    else {
                        _this.pelcod.focusNear(false);
                        _this.pelcod.focusFar(false);
                    }
                    _this.pelcod.send();
                }
            }`
If I send focusNear command then data.value === 0 .
I changed  if (data.value < 0) to  if (data.value <= 0).

I have other question about config in the rposConfig.json 
`"PTZOutputURL": "127.0.0.1:9999",`
What does this line mean? 
Can I use Raspbery pi to relay onvif commands to another onvif camera?
For example: raspberry has the address 192.168.1.2. 
There is a camera on the local network with the address 192.168.1.3. 
My videoserver sends the zoom command to the address 192.168.1.2. 
Raspberry recieve this command, sends it to the serial port and sends the onvif command to the camera via TCP.
The camera also responds to this command.

RogerHardiman, 2020-02-05:
I have other question about config in the rposConfig.json
"PTZOutputURL": "127.0.0.1:9999",
What does this line mean?

RPOS will receive PTZ commands in ONVIF Format.
It can convert the PTZ Command into older 'serial data' PTZ format (sometimes called RS485 data). The serial data formats supported are Pelco-D, Pelco-P and Sony VISCA.
RPOS then has to send the Serial Data (eg Peldo-D serial data) somewhere.
It can send it to a USB COM Port or it can send it to a Socket. In this case 127.0.0.1:9999 is sending the Pelco-D serial data to a Socket.
There would be some other server listening on Port 9999 that knows what to do with Pelco-D serial data.

It does not send an ONVIF PTZ command on to the URL. That could be added. But right now it is just RS485 Serial Data for Analogue CCTV cameras.

RogerHardiman, 2020-02-05:
Hi

The ONVIF Standard says that Positive speeds are one direction for Focus and Negative Speeds are the other direction for focus. There is nothing in there for zero values.
There is an official FocusStop command however it may be that some VMS systems use a Focus with speed 0 to stop the focus command.

So could you see if Milesight outputs Focus Stop Correctly?
I'll try and test it with a few other VMS systems and check the output Focus-Stop and then we can make this change.

Roger

RogerHardiman, 2020-02-05:
Hi
Thanks for the bug report about the focus.

I found the cause of the problem.
There was a bug in RPOS with the GetMoveOptions command.
It returns a range of valid speeds and the range was set from 0.0 to 1.0.
Milesight was doing the right thing and capping the output to this range.

GetMoveOptions should have returned a range of -1.0 to +1.0.
With this fix, Milesight works properly.

RogerHardiman, 2020-02-05:
Fix is now in the code. See this commit...

https://github.com/BreeeZe/rpos/commit/9b7920419f021f9d4c0a85709fbce42ecc8017d3

closed 2020-02-05 by RogerHardiman