BreeeZe/rpos #83: Node version issue

JoelATX, 2020-03-08:
I think i'm really close to having it working. Still can't get my server to see the server.
I installed node 8.
Here is what I'm getting. Sorry it's kinda long.

pi@raspberrypi:~ $ cd rpos
pi@raspberrypi:~/rpos $ node rpos.js
execSync('cat /proc/device-tree/model')
execSync('cat /proc/cpuinfo')
Read IP address from awdl0 failed
Read IP address 192.168.1.200 from eth0
Manufacturer : RPOS Raspberry Pi
Model : Model_B+_PI_3
HardwareId : 
SerialNumber : 0000000039017b62
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.1.200:8081/
execSync('v4l2-ctl --set-fmt-video=pixelformat=4')
execSync('v4l2-ctl --set-fmt-video=width=1280,height=720')
execSync('v4l2-ctl --set-parm=25')
execSync('v4l2-ctl --set-priority=3')
execSync('v4l2-ctl --set-ctrl power_line_frequency=3')
execSync('v4l2-ctl -l')
Controlvalue 'brightness' : 50
Controlvalue 'contrast' : 0
Controlvalue 'saturation' : 0
Controlvalue 'red_balance' : 1000
Controlvalue 'blue_balance' : 1000
Controlvalue 'horizontal_flip' : 0
Controlvalue 'vertical_flip' : 0
Controlvalue 'power_line_frequency' : 3
Controlvalue 'sharpness' : 0
Controlvalue 'color_effects' : 0
Controlvalue 'rotate' : 0
Controlvalue 'color_effects_cbcr' : 32896
Controlvalue 'video_bitrate_mode' : 0
Controlvalue 'video_bitrate' : 10000000
Controlvalue 'repeat_sequence_header' : 0
Controlvalue 'h264_i_frame_period' : 60
Controlvalue 'h264_level' : 11
Controlvalue 'h264_profile' : 4
Controlvalue 'auto_exposure' : 0
Controlvalue 'exposure_time_absolute' : 1000
Controlvalue 'exposure_dynamic_framerate' : 0
Controlvalue 'auto_exposure_bias' : 12
Controlvalue 'white_balance_auto_preset' : 1
Controlvalue 'image_stabilization' : 0
Controlvalue 'iso_sensitivity' : 0
Controlvalue 'exposure_metering_mode' : 0
Controlvalue 'scene_mode' : 0
Controlvalue 'compression_quality' : 30
Binding DeviceService to http://192.168.1.200:8081/onvif/device_service
Binding MediaService to http://192.168.1.200:8081/onvif/media_service
Binding PTZService to http://192.168.1.200:8081/onvif/ptz_service
Binding ImagingService to http://192.168.1.200:8081/onvif/imaging_service
discovery_service started
ptz_service started
device_service started
imaging_service started
Starting Live555 rtsp server
spawn('v4l2rtspserver', [-P,8554,-u,h264,-m,h264m,-M,224.0.0.1:10001,-W,1280,-H,720,/dev/video0], undefined)
media_service started
rtspServer: log level:500

web request received : /
execSync('v4l2-ctl -l')
Controlvalue 'brightness' : 50
Controlvalue 'contrast' : 0
Controlvalue 'saturation' : 0
Controlvalue 'red_balance' : 1000
Controlvalue 'blue_balance' : 1000
Controlvalue 'horizontal_flip' : 0
Controlvalue 'vertical_flip' : 0
Controlvalue 'power_line_frequency' : 3
Controlvalue 'sharpness' : 0
Controlvalue 'color_effects' : 0
Controlvalue 'rotate' : 0
Controlvalue 'color_effects_cbcr' : 32896
Controlvalue 'video_bitrate_mode' : 0
Controlvalue 'video_bitrate' : 10000000
Controlvalue 'repeat_sequence_header' : 0
Controlvalue 'h264_i_frame_period' : 60
Controlvalue 'h264_level' : 11
Controlvalue 'h264_profile' : 4
Controlvalue 'auto_exposure' : 0
Controlvalue 'exposure_time_absolute' : 1000
Controlvalue 'exposure_dynamic_framerate' : 0
Controlvalue 'auto_exposure_bias' : 12
Controlvalue 'white_balance_auto_preset' : 1
Controlvalue 'image_stabilization' : 0
Controlvalue 'iso_sensitivity' : 0
Controlvalue 'exposure_metering_mode' : 0
Controlvalue 'scene_mode' : 0
Controlvalue 'compression_quality' : 30
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received
Discovery received

RogerHardiman, 2020-03-14:
so what happens when you browse to        
   http://192.168.1.200:8081/

RogerHardiman, 2020-03-14:
also what camera viewing software are you trying

JoelATX, 2020-03-14:
Hi Roger,

I purchased the Camect ( www.camect.com <http://www.camect.com/> )server from it’s kickstarter campaign.

After making the post, I continued to look thru more posts and found what the streaming URL should look like. ( 192.168.1.200/h264 )  Using that stream I was able to get the server to record the stream. After more rooting around it looked like changing the stream to 1080 would not be possible. I paused my search.

I felt like I was close, because the Camect server was sending the discovery request and the your software was trying to respond. That’s when I made the post.
Camect’s developer said it doesn’t really keep a log of the ONVIF requests/responces. Because my questions are WAY out of bounds for normal tech support, I didn’t push the request.

Some of the depended software installs were kinda murky for me. I wasn’t sure if BreeeZe would work with all current installs. I’m just a monkey pounding keys until something happens.

What I’m trying to do  - I have some lizards, Red Eyed Crocodile skinks. They look like a crocodile with a Night Fury head from "How to Train your Dragon.”  I’m looking for cameras that I can put close focusing lenses on that are under $100. Your software would be perfect for what I’m trying to do.

Currently the Pi is running motionEye, serving stream to the Camect. I can also log into motionEye.

I put BreeeZe on a separate SD card from the motionEye install. So I’m willing the work with it more. If you can make recommendations. I know how these projects  can be all consuming for a while, not sure if you still have time for it in your world.

Thanks for any help you can give.

Joel

JoelATX, 2020-03-14:
I really didn’t answer this…

I was able to make settings changes to the camera. I think, if this is the URL for settings I could make settings changes.

RogerHardiman, 2020-03-15:
I've had a look at camect and also at the log you posted.

There are two main ways to talk to a CCTV Camera.
You can go direct to the RTSP video stream and you've figured out the RTSP URL which is great.

The other way is ONVIF.
ONVIF Comes in two parts
Part 1 is discovering the device on the local LAN. We can see in the Logs you poste that RPOS receives the discovery message and sends a reply.

But then we need to check if RPOS actually gets any ONVIF commands.
There are two RPOS settings, logLevel and logSoapCalls.
Change them to this and let's see if camect is sending any ONVIF to RPOS.

So
  "logLevel" : 4, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
  "logSoapCalls" : true

Finally there may be a way in Camect to add ONVIF cameras that are not autodetected.
If you try that, the ONVIF URL is
http://192.168.1.200:8081/onvif/device_service

(note the Port 8081 in there, that's important)

Roger

RogerHardiman, 2020-05-11:
did you change the log level and logSoapCalls and capture the logs?

JoelATX, 2020-05-11:
Hi,

Thanks for checking.

Sorry No, not yet. Too many other pressing (and boring) projects over the last two months.

I had to stop myself from erasing the SD card.

I haven’t had any CoronaTime off, I work in printing.

Joel

RogerHardiman, 2020-05-11:
Hi.
No problem. Drop me a line via this Issue Report when you can.