BreeeZe/rpos #84: Live view and stream

briansonline, 2020-03-18:
Hi I'm a bit of noob but I've managed to get this working really well and have had a play around with it but I'm wondering if there's a way to preview the raw output on local hdmi and simultaneously streaming, or an onvif viewer to achieve the same result. I have lots of other questions and a couple of different applications for this. Is there a way I can contact you? Thanks. Brian

RogerHardiman, 2020-03-18:
I've not thought about outputtingon HDMI before.
One way, not ideal, would be to run omxplayer and connect to the local RTSP stream, eg rtsp://127.0.0.1/h264
If you use the gstreamer mode in RPOS there is probably a way to output vide in the gstreamer pipeline

You can get my email from the bottom of www.rjh.org.uk
(sorry for the extra stop, just avoiding email harvesters and spam)

RogerHardiman, 2020-05-11:
Did you try the suggestion I had sent you to remove preview=false from the Python script?

briansonline, 2020-05-11:
Hiya buddy, sorry for the delay.

I’ve been pretty busy and to be honest I did get sidetracked with 3d printing and trying to get the thermal and face detection working again!

I’ve managed to get rpos working again but I’m a bit rusty. I have changed the only line in the file with preview to preview=true and tried with 2 hdmi monitors connected, but it’s not using the Gstreamer option and it doesn’t seem to be in the rposConfig.json file either (only options 1 and 2), so I’ve kinda given up for now! I’ll have another mess around though if you have it working.

Cheers
Speak soon
Brian