BreeeZe/rpos #85: Uncaught Exception

erdose, 2020-04-11:
Hi,

I need some help!
I followed the ReadMe's instructions step-by-step but I got an exception in the last step.

> pi@raspberrypi:~/rpos $ node rpos.js
> Read IP address 192.168.1.79 from wlan0
> Manufacturer : RPOS Raspberry Pi
> Model : Model_B_PI_3
> HardwareId :
> SerialNumber : 000000009d2e177b
> FirmwareVersion : 2.0.4
> Starting camera settings webserver on http://192.168.1.79:88/
> Uncaught Exception... : Error: ENOENT: no such file or directory, chmod './bin/rtspServer'
>     at Object.chmodSync (fs.js:1023:3)
>     at new Camera (/home/pi/rpos/lib/camera.js:75:16)
>     at Object.<anonymous> (/home/pi/rpos/rpos.js:39:14)
>     at Module._compile (internal/modules/cjs/loader.js:689:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
>     at Module.load (internal/modules/cjs/loader.js:599:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
>     at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
>     at startup (internal/bootstrap/node.js:283:19)

Thanks!

bespiloy, 2020-04-16:
I have same problem. Anyone have fix? thank you in advance.

pi@rpi1101:~/rpos $ node rpos.js
Read IP address 192.168.100.49 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_Revision_2.0
HardwareId :
SerialNumber : 000000008c4ed4a8
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.100.49:8081/
Uncaught Exception... : Error: ENOENT: no such file or directory, chmod './bin/rtspServer'
    at Object.chmodSync (fs.js:1023:3)
    at new Camera (/home/pi/rpos/lib/camera.js:75:16)
    at Object.<anonymous> (/home/pi/rpos/rpos.js:39:14)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:283:19)

RogerHardiman, 2020-04-16:
I'lll take a look

RogerHardiman, 2020-04-16:
It seems the bin/rtspServer file got accidentally deleted in a recent Pull Request for USB Camera support with GStreamer video.

I've restored the file and pushed the change to Github. So if you get the latest source from github it should be working again. (I've just done a fresh install to check)

erdose, 2020-04-16:
Hi Roger,

thanks to the fix, it's already okay.
But I have another problem. (Please check the attached picture.)
What will be the problem?
[image]
[image]

bespiloy, 2020-04-16:
Thank you Roger. I redo the step # 3, 4.1.a and 6. Last I did the "node rpos.js". It went well but I'm not sure what should I do next. Below are the command lines and output I got from RPI console/terminal:

pi@rpi1101:~/rpos $ node rpos.js
Read IP address 192.168.100.49 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_Revision_2.0
HardwareId :
SerialNumber : 000000008c4ed4a8
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.100.49:8081/
v4l2ctl.json does not exist yet or invalid.
Binding DeviceService to http://192.168.100.49:8081/onvif/device_service
Binding MediaService to http://192.168.100.49:8081/onvif/media_service
Binding PTZService to http://192.168.100.49:8081/onvif/ptz_service
Binding ImagingService to http://192.168.100.49:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started
^CCtrl-C...
Stopping Live555 rtsp server
pi@rpi1101:~/rpos $ node rpos.js
Read IP address 192.168.100.49 from wlan0
Manufacturer : RPOS Raspberry Pi
Model : Model_B_Revision_2.0
HardwareId :
SerialNumber : 000000008c4ed4a8
FirmwareVersion : 2.0.4
Starting camera settings webserver on http://192.168.100.49:8081/
Binding DeviceService to http://192.168.100.49:8081/onvif/device_service
Binding MediaService to http://192.168.100.49:8081/onvif/media_service
Binding PTZService to http://192.168.100.49:8081/onvif/ptz_service
Binding ImagingService to http://192.168.100.49:8081/onvif/imaging_service
discovery_service started
imaging_service started
ptz_service started
device_service started
Starting Live555 rtsp server
media_service started

From erdose screenshot, after "media_service started" output message, the "rtspServer: Streaming on URL rtsp://192.168.1.69:8564/h264" appeared but nothing from mine. What do you think is missing? thank you.

RogerHardiman, 2020-04-17:
Hi @erdose 

Thanks for the screen shots.

ONVIF Protocol allows a CCTV client to request a JPEG image from the camera. (this is not mjpeg streaming video - this is just a single frame of video)
Because RPOS can use the V4L2 API to talk to the camera then getting a single frame was not possible when the camera is open in 'live video' mode.
So I uses ffmpeg to open a rtsp connection to 127.0.0.1 (localhost) to get a still image out of the live video steam. It is a hack but the best I could do.

So first you need to make sure ffmpeg is installed.

But there is also a whole load of bugs you have uncovered.

1) I noticed you started your RTSP server on port 8564 but ffmpeg was trying to connect to 127.0.0.1 port 8554.  This is an RPOS bug.
I need to get RPOS ffmpeg to connect to the right port.
THIS IS FIXED

2) I also found that if you change the RTSPName from h264 to something else like 'h264Stream1' then it is fine with RTSP Server #2, the one from mpromonet, but not with the basic RTSP server as that is hard coded to a name of /h264

Anyway I've made the right changes to RPOS.

Roger

erdose, 2020-04-18:
Hi @RogerHardiman 

thanks for the fix but the problem is still unchanged.
I checked the FFmpeg, but it seems to be OK.

> pi@raspberrypi:~ $ which ffmpeg
> /usr/local/bin/ffmpeg

I didn't do the whole install process from the beginning, I just repeat these steps:
`cd rpos` `git pull` `npm install` `npx gulp` and the final step `node rpos.js`
Is it enough or do I need to reinstall the full image and packages from scratch?

Otherwise, I want to use it with Synology NAS (Surveillance Station 8.2) and the connection and configuration are already okay but the stream is weak (FPS: 0). Please check the attached pictures!

[image]
[image]

Thanks!

Other information:
rposConfig.json

> {
>   "NetworkAdapters" : ["awdl0","eth0", "wlan0", "en0"],
>   "IpAddress" : "192.168.1.69",
>   "ServicePort" : 8081,
>   "Username" : "test",
>   "Password" : "test",
>   "CameraType" : "picam",
>   "RTSPAddress" : "", "//":"Normally left blank. Used to set RTSP Server Address",
>   "RTSPPort" : 8566,
>   "RTSPName" : "h264",
>   "MulticastEnabled"  : false,
>   "RTSPMulticastName" : "h264m",
>   "MulticastAddress"  : "224.0.0.1",
>   "MulticastPort"     : "10001",
>   "RTSPServer" : 1, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",
>   "PTZDriver" : "none", "PTZDriverComment": "## valid values are none,tenx,pelcod,visca and pan-tilt-hat",
>   "PTZOutput" : "none", "PTZOutputComment": "## values are none (eg Tenx), serial and tcp",
>   "PTZSerialPort" : "/dev/ttyUSB0",
>   "PTZSerialPortSettings" : { "baudRate":2400, "dataBits":8, "parity":"none", "stopBits":1 },
>   "PTZOutputURL": "127.0.0.1:9999",
>   "PTZCameraAddress": 1,
>   "DeviceInformation" : {
>     "Manufacturer" : "Raspberry Pi",
>     "Model" : "2 B",
>     "HardwareId" : ""
>   },
>   "logLevel" : 3, "logLevelComment": "## LogLevels are > 1:Error 2:Warning 3:Info 4:Debug",
>   "logSoapCalls" : false
> }

RogerHardiman, 2020-04-26:
Not used synology before.
Have yiu tried the 2nd RTSP server option - the one from mpromonet.  That is what I use.

erdose, 2020-04-26:
Hi @RogerHardiman 

Thank you very much! :)
It works well!

RogerHardiman, 2020-04-29:
Let's close it but I'll change the README to say don't bother with Option 1 for the RTSP server

closed 2020-04-29 by RogerHardiman