BreeeZe/rpos #86: rpos执行npx gulp报错

fangjiechunv, 2020-04-25:
hardware：Raspberry Pi 3B+
os：2018-11-13-raspbian-stretch-full

**error1：**
`(node:5866) UnhandledPromiseRejectionWarning: TypingsError: Unable to read typings for "body-parser". You should check the entry paths in "body-parser.d.ts" are up to date`
**error2：**
```
(node:5866) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)
(node:5866) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
  Node version 6 will prompt error 1, node version 8 will prompt error 1 and error 2, the result is that the .ts file cannot be turned into a .js file, because no rpos.js is generated, so the next step cannot be performed. The network can not find an effective way to deal with errors, so I can only ask questions, I hope to take the time to answer, thank you.
ps: The entire operation process is strictly in accordance with the process on GitHub, using nvm to switch versions.

RogerHardiman, 2020-04-26:
Hi
In run stretch on my Pi and node v8. It all runs just fine and I checked out a clean copy of rpos to check.
So I am not sure what is causing your problem.

fangjiechunv, 2020-04-27:
thank you. I have tried all kinds of tutorials on the Internet, and I have not been able to successfully compile .ts files. I also considered the system version and other issues, so every time I backup and restore the system, the result is that the problem still exists. I also tried to use the git git reset --hard command to check out the old version of the project to compile, but it still does not work. Almost desperate. Probably bad luck.

RogerHardiman, 2020-04-27:
Do you have a spare SD card?
Can you take a fresh install of raspbian and try that please. Maybe you have a globally installed package that conflicts.

Then we can check that we can get it working on a spare SD card and maybe look back at our existing OS

fangjiechunv, 2020-04-27:
Thank you again for your patience. I don't have any extra sd cards. But in order to ensure that it is not a system or other environmental conflict, I directly backed up the image of the sd card, then flashed the sd card into a new system, and followed the GitHub steps to test from scratch. Unfortunately, it has been stopped compiling ts files here.

RogerHardiman, 2020-04-27:
I was going to try and install the same rasbian as you on a spare SD card.
I have two questions
1) Your initial post said you were using 2018-11-13-raspbian-stretch-full

I can see the full image at http://downloads.raspberrypi.org/raspbian/images/raspbian-2018-11-15/  Is that the right image?

2) Are you fully up to date with sudo apt update and sudo apt upgrade so I can get to the same OS revision as you?

Thanks

fangjiechunv, 2020-04-27:
Thank you again for taking the time to answer.
Answer your questions first:
1. I downloaded the full version of the system, URL: https://downloads.raspberrypi.org/raspbian_full/images/raspbian_full-2018-11-15/2018-11-13-raspbian-stretch-full.zip
2. Because I am in China, in order to improve the update speed, I changed the source to Alibaba source or Tsinghua source, but after the modification, sudo apt-get update and sudo apt-get upgrade are executed.

In addition, in fact, I started with the latest version of Raspberry Pi OS, such as 2020-02-13-raspbian-buster, but various attempts could not be compiled, and I began to wonder if it was an OS problem, so I tried to replace the old one The version of the system, such as the stretch version mentioned at the beginning, but this old version will not work. So in the end it can basically be ruled out as a system problem. So I suggest that you do n’t have to waste time on replacing the old system, it should be other reasons.

RogerHardiman, 2020-05-11:
I installed the latest Raspbian Buster with Desktop on my Pi 3B+ a few days ago.
(It was not the 'With Desktop and Recommended Software)

The installation of RPOS worked just fine.