BreeeZe/rpos #87: Feature request: Edge Recording

erdose, 2020-04-28:
Hi,

Thanks for the work guys it is really cool but I'd like to ask for a useful new feature, the **Edge Recording** if it is possible. As far as I know, this is part of the Profile G standard. For example, the Synology NAS supports this feature. [https://www.synology.com/en-global/knowledgebase/Surveillance/tutorial/Camera/How_to_set_up_edge_recording_in_Surveillance_Station](url)
Could someone help me?
Thanks a lot!

RogerHardiman, 2020-04-29:
It needs a volunteer to write the software.
The we can review it, test it and add it to RPOS.

NiCyph, 2020-06-22:
Hi - I would also like to have onvif profile g support so that I can do edge recording instead of all the processing on my synology NAS.
Sorry, I can't write software but happy to help testing.
Thanks.