BreeeZe/rpos #96: Issues running rpos

magallardo, 2020-09-30:
Hello, I have followed the instructions and I have partially got rpos to work.
I installed on a RPi4 with 4 gig of memory running Raspberry PI OS version: August 2020.
As mentioned previously, the install seems to be ok but when I try to access the camera using Onvif Device Manager, I get snapshots of the camera when I click Live Video or Video Streaming.  Right after the snapshot is displayed, Onvif device manager shows a No Signal window.  If I click again, I get a new snapshot but not the live video.
I also tried using a golang onvif library and I am able to communicate with the camera but I only get a black image snapshot.
Has anybody advise on what could be happening?  I am attaching the logs from rpos.  It shows some errors about some library version incompatibilities.  Please advise.
[RposIssue.log](https://github.com/BreeeZe/rpos/files/5306976/RposIssue.log)

tal0301, 2020-11-04:
Have you managed to fix it? I'm having the same problem...

Error 5xx for the RTSP Server

uglurass, 2020-11-04:
Trying older versions might fix you temporarily until að fix has been
implemented!

tal0301, 2020-11-04:
> Trying older versions might fix you temporarily until að fix has been implemented!
> […](#)

Any specific suggestion or should I just try one and see how it goes?

aicjofs, 2020-11-26:
Are you using rpicam or a usb camera?  Are you using gstreamer apt package or building from source?  Which template did you use for rposConfig.json?

I used a usb camera with the usb rposConfig template, and the Gstreamer apt-get package.  This seemed to work.   

I used a picam with the picam rposConfig template, and the Gstreamer apt-get package.  This did not work.  Onvif Device Manager will give a brief snapshot and then no signal.  I tried changing the rtsp server in the config to 3 to use Gstreamer 

`"RTSPServer" : 1, "RtspServerComment" : "## Select RTSP Server > 1:RPOS RTSP Server 2:V4L2 RTSP Server by mpromonet (auto selected if MulticastEnabled=true)",`

That worked but periodically would stop with no signal, maybe this will fix it for you but it wasn't a final solution in my setup.

Built gstreamer 1.18 from source and left the above config for RTSPServer at "3" and it works.

You need meson and ninja, and possibly some other dependencies.  
https://github.com/GStreamer/gst-build

Karlo318, 2022-03-27:
Has anyone managed to get rpos running on Raspberry Pi Zero? I am facing the same problem and no matter what I do, I always end up with NO SIGNAL message.