BreeeZe/rpos #98: Inability to use v4l2loopback

tal0301, 2020-10-18:
Hi All,

I need to stream a video file using ONVIF, and I'm trying to use v4l2loopback. It creates an emulated device that works and shows video, but the rpos cannot pick it up (showing no video on the ONVIF device Manager).

I'm sure that I'm missing something in the Gstreamer Pipeline, but I cannot figure it out.

Can someone please help?

Thank you.

bagrintech, 2020-10-19:
I have done it with vlc rtsp server.

tal0301, 2020-10-19:
Any chance to share with me your insights? I'm trying to work around this
issue...

RogerHardiman, 2020-10-19:
vlc has a built in RTSP server.
So you can start VLC from the command line to make it read a file (with looping over and over) and pass on the video via its built in RTSP server. 
I searched around on the internet for the command line syntax.

You would then use the VLC RTSP server instead of the Live555 based RTSP server or the GStreamer RTSP Server

bagrintech, 2020-10-19:
Get root access for VLC: sudo sed -i 's/geteuid/getppid/' /usr/bin/vlc
Build py script :  
`subprocess.Popen(["vlc","-I","dummy","--loop","/home/bagrin/video.mp4",":sout=#gather:rtp{sdp=rtsp://192.168.1.3:8554/h264}",":network-caching=500",":sout-all",":sout-keep"], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE, text=True)`
Starting py script